chipsalliance/chisel #2470: bug: should be able to instantiate a memory in a RawModule

ekiwi, 2022-04-01:
**Type of issue**: bug report

It used to be possible to instantiate a `Mem` or `SyncReadMem` in  `RawModule` without an implicit clock. The 3.5 release broke this.
Now you get a somewhat cryptic error message:

```
chisel3.internal.ChiselException: Error: No implicit clock.
```

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Example from a real chisel project: https://scastie.scala-lang.org/zGQEY02JTBO3zYEtz6Mh5Q

**What is the current behavior?**
Cryptic error message, then elaboration fails:

```
chisel3.internal.ChiselException: Error: No implicit clock.
```

**What is the expected behavior?**

The circuit should successfully elaborate as was the case with older Chisel releases.

ekiwi, 2022-04-01:
This regression was most likely introduced by: https://github.com/chipsalliance/chisel3/pull/2333

adkian-sifive, 2022-04-01:
> It used to be possible to instantiate a Mem or SyncReadMem without an implicit clock

Was that actually a good thing? Conceptually, I'm not sure whether memories are valid without a clock associated with them?

mwachs5, 2022-04-01:
agreed... similar to a Reg, what does it even mean to have a Mem or SyncReadMem without a clock?

ekiwi, 2022-04-01:
This is the original source of my example, btw: https://github.com/hansemandse/KWSonSNN/blob/master/src/main/scala/neuroproc/TrueDualPortMemory.scala

mwachs5, 2022-04-01:
So maybe this should have been an Option: https://github.com/chipsalliance/chisel3/pull/2333/files#diff-69a38113a9a1186347fec4aa8cdd4b0366b5bac9d5eb4aca99fcf98717844bf4R59

and if it fails we use a "None", then the warning can say "None" was the implied clock...

seldridge, 2022-04-01:
👍  On making that an `Option[Clock]`. It's reasonable to represent memories this way as clock crossings and the current API doesn't allow you to avoid the warning in that situation.

ekiwi, 2022-04-01:
> So maybe this should have been an Option: https://github.com/chipsalliance/chisel3/pull/2333/files#diff-69a38113a9a1186347fec4aa8cdd4b0366b5bac9d5eb4aca99fcf98717844bf4R59
> 
> and if it fails we use a "None", then the warning can say "None" was the implied clock...

Yes. Or maybe just do not emit the warning if there is not implicit clock. If someone chooses to use a `RawModule` they probably know what they are doing since they will have to create clock scopes explicitly.

mwachs5, 2022-04-01:
My suggestion was to still emit the warning. Is it possible to write this in a way that avoids the warning?

ekiwi, 2022-04-01:
> My suggestion was to still emit the warning. Is it possible to then re-write this circuit in a way that avoids the warning?

I guess we could allow this to be written like:

```.scala
class TrueDualPortMemory(addrW: Int, dataW: Int) extends RawModule {
  val io = IO(new TrueDualPortMemoryIO(addrW, dataW))
  val ram = SyncReadMem(1 << addrW, UInt(dataW.W))

  // Port a
  io.doa := DontCare
  when(io.ena) {
    when(io.wea) {
      ram(io.addra, io.clka) := io.dia
    }
    io.doa := ram(io.addra, io.clka)
  }

  // Port b
  io.dob := DontCare
  when(io.enb) {
    when(io.web) {
      ram(io.addrb, io.clka) := io.dib
    }
    io.dob := ram(io.addrb, io.clka)
  }
}
```

That way it would not be more complicated than the original and the clock would be more explicit.

adkian-sifive, 2022-04-01:
Yeah this compiles with no warnings with a workaround: https://scastie.scala-lang.org/bAYbdYWkQEKQm9H9Q5FhMA

I'm guessing you want `io.clkb` for the "port b"?

ekiwi, 2022-04-01:
> Yeah this compiles with no warnings with a workaround: https://scastie.scala-lang.org/bAYbdYWkQEKQm9H9Q5FhMA

What I meant was that you should not have to declare the clock twice. So you could just specify the clock for the port instead of having to make a clock scope.

I don't think having to make a clock scope and also to declare the clock at the port is a good API.

commit 44165a25 (2022-04-19): Allow creating memories without an implicit clock (#2494)

Fixes #2470
commit 70da39e1 (2022-04-19): Allow creating memories without an implicit clock (#2494) (#2495)

Fixes #2470

(cherry picked from commit 44165a259bb16733a41798edca6b554b13f1d54a)
closed 2022-04-19 by jackkoenig