chipsalliance/chisel #2508: Connecting a DontCare to a DataView Produces Illegal FIRRTL

seldridge, 2022-04-26:
Consider the following:

```scala
import chisel3._
import chisel3.experimental.dataview._

class Foo extends Bundle {
  val foo = UInt(8.W)
}
class Bar extends Foo {
  val bar = UInt(8.W)
}
class MyModule extends Module {
  val foo = IO(Input(new Foo))
  val bar = IO(Output(new Bar))
  bar.viewAsSupertype(new Foo) := DontCare // bar.foo := foo.foo
  bar.bar := 123.U           // all fields need to be connected
}
```

This produces the following FIRRTL IR which is illegal:
```scala
circuit MyModule :
  module MyModule :
    input clock : Clock
    input reset : UInt<1>
    input foo : { foo : UInt<8>}
    output bar : { foo : UInt<8>, bar : UInt<8>}

    view.foo is invalid @[main.scala 15:32]
    bar.bar <= UInt<7>("h7b") @[main.scala 16:11]
```

Specifically `view` is never declared. 😬 

A scastie snippet of the above is here: https://scastie.scala-lang.org/seldridge/WcJpoy9RTEOTLGQGGjUbiA/10

This can be worked around with a `WireInit(new Foo, DontCare)`.  However, it is surprising that the user needs to do this.

terpstra, 2022-04-27:
Glad this is being fixed. Posting so I get notified of progress. :)

mwachs5, 2022-04-27:
@seldridge can you clarify the workaround (perhaps with another scastie)?

jackkoenig, 2022-05-18:
You can more easily work around this by just using `<>` (https://scastie.scala-lang.org/ypfeMJGnQ4iVOz3YcAO58w).
Fix coming.

closed 2022-05-19 by jackkoenig