chipsalliance/chisel #2510: Result of operator ## cannot be assigned to

forflo, 2022-04-29:
**Type of issue**: bug report | feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: proposal

**What is the current behavior?**

Assume two kinds of Blackbox modules, `Slave` and `Master`. Let their interfaces be denoted by `SlaveIf` and `MasterIf`. Now, consider the case where you want to connect two `Master` instances to one of the `Slave` instances like in the following code:

```scala
class SlaveIf extends Bundle {
  val inp = Input(UInt(4.W))
  val outp = Output(UInt(4.W))
}

class MasterIf extends Bundle {
  val inp = Output(UInt(2.W))
  val outp = Input(UInt(2.W))
}

class Slave extends BlackBox {
  val io = IO(new SlaveIf)
}

class Master extends BlackBox {
  val io = IO(new MasterIf)
}

class Tlf extends Module {
  val io = IO(new Bundle {
    val fnord = Input(UInt(4.W))
  })

  val xbar = Module(new Slave)
  val mst1 = Module(new Master)
  val mst2 = Module(new Master)

  xbar.io.inp <> (mst1.io.inp ## mst2.io.inp) // works
  xbar.io.outp <> (mst1.io.outp ## mst1.io.outp) // !!!!not working!!!!
}
```

`xbar.io.inp <> (mst1.io.inp ## mst2.io.inp)` will roughly produce `assign xbar_inp = {mst1_inp,mst2_inp};` in the Verilog file, but the last line will produce the following runtime exception

```
Exception in thread "main" chisel3.internal.ChiselException: internal error: attempted to generate LHS ref to ReadOnlyBinding OpBinding(chisel_test.Tlf@1dbb650b,None)
        at ... ()
        at chisel_test.Tlf.<init>(App.scala:38)
        at chisel_test.App$.$anonfun$main$1(App.scala:46)
        at ... ()
        at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
```

Essentially it says that the result of a `##` operator is not writable.

This is overly conservative. Sure, the result of the `+` Operator on UInt cannot be written to, but the result of `##` can be.

**What is the expected behavior?**

The code should produce three module instances, some wires and the following two assignments

```
assign xbar_inp = {mst1_inp,mst2_inp};
assign {mst1_outp, mst2_outp} = xbar_outp
```

**Environment:**

- Chisel version: 3.5.2
- Latest macOS

ekiwi, 2022-04-29:
Chisel does not support sub-word assignments. There is no principled reason for not supporting it besides the fact that no one is currently working on it.

I recently tried to add support for sub-word assignments to our compiler. It turned out to be more than a 2-3 day weekend project and work has kept me from completing it. https://github.com/ekiwi/firrtl/tree/sub-word-assign

forflo, 2022-04-29:
> I recently tried to add support for sub-word assignments to our compiler. It turned out to be more than a 2-3 day weekend project and work has kept me from completing it. https://github.com/ekiwi/firrtl/tree/sub-word-assign

Nice! Would be glad to replace my current workaround with the expression above :)