chipsalliance/chisel #2531: loadMemoryFromFileInline can not generate readmem in generated verilog file

JACKLIAO0, 2022-05-14:
**Type of issue**: bug report

**information**
 

I have generated a SRAM on cbus : 
`trait HasTestRAM { this: BaseSubsystem =>
  val testRAM = LazyModule(
    new TLRAM(AddressSet(0x40000000, 0x7FFFF), beatBytes = cbus.beatBytes)
  )
  testRAM.node := cbus.coupleTo("sram") { TLFragmenter(cbus.beatBytes,4096) := _ }
}`
and then put HasTestRAM in ExampleRocketSystem.

I want to use loadMemoryFromFileInline  API to load from data file to SRAM like this :

`loadMemoryFromFileInline(ldut.testRAM.module.mem, "xxx.txt")` in Testharess.scala 

In generated verilog, I can't find ”readmemh“ . Is there something wrong with my way of using this loadMemoryFromFileInline ?
@sequencer

JACKLIAO0, 2022-05-15:
I try to run a simplest demo to test whether loadMemoryFromFileInline can work correctly like following:
  `import chisel3._
import chisel3.stage.ChiselStage
import chisel3.util.experimental.loadMemoryFromFileInline

class InitMemInline(memoryFile: String = "") extends Module {
  val width: Int = 32
  val io = IO(new Bundle {
    val enable = Input(Bool())
    val write = Input(Bool())
    val addr = Input(UInt(10.W))
    val dataIn = Input(UInt(width.W))
    val dataOut = Output(UInt(width.W))
  })

  val mem = SyncReadMem(1024, UInt(width.W))
  // Initialize memory
  if (memoryFile.trim().nonEmpty) {
    loadMemoryFromFileInline(mem, memoryFile)
  }
  io.dataOut := DontCare
  when(io.enable) {
    val rdwrPort = mem(io.addr)
    when (io.write) { rdwrPort := io.dataIn }
      .otherwise    { io.dataOut := rdwrPort }
  }
}

object myRtlEmitter extends App {
  val prefix = "xxxx"

  (new ChiselStage).emitVerilog(
    new InitMemInline,
    args ++ Array("-td", "./build/test"),
    )
}`

**but the generated verilog have still no readmem,like this: **
`module InitMemInline(
  input         clock,
  input         reset,
  input         io_enable,
  input         io_write,
  input  [9:0]  io_addr,
  input  [31:0] io_dataIn,
  output [31:0] io_dataOut
);
`ifdef RANDOMIZE_MEM_INIT
  reg [31:0] _RAND_0;
`endif // RANDOMIZE_MEM_INIT
`ifdef RANDOMIZE_REG_INIT
  reg [31:0] _RAND_1;
  reg [31:0] _RAND_2;
`endif // RANDOMIZE_REG_INIT
  reg [31:0] mem [0:1023]; // @[memtest.scala 19:24]
  wire  mem_rdwrPort_r_en; // @[memtest.scala 19:24]
  wire [9:0] mem_rdwrPort_r_addr; // @[memtest.scala 19:24]
  wire [31:0] mem_rdwrPort_r_data; // @[memtest.scala 19:24]
  wire [31:0] mem_rdwrPort_w_data; // @[memtest.scala 19:24]
  wire [9:0] mem_rdwrPort_w_addr; // @[memtest.scala 19:24]
  wire  mem_rdwrPort_w_mask; // @[memtest.scala 19:24]
  wire  mem_rdwrPort_w_en; // @[memtest.scala 19:24]
  reg  mem_rdwrPort_r_en_pipe_0;
  reg [9:0] mem_rdwrPort_r_addr_pipe_0;
  assign mem_rdwrPort_r_en = mem_rdwrPort_r_en_pipe_0;
  assign mem_rdwrPort_r_addr = mem_rdwrPort_r_addr_pipe_0;
  assign mem_rdwrPort_r_data = mem[mem_rdwrPort_r_addr]; // @[memtest.scala 19:24]
  assign mem_rdwrPort_w_data = io_dataIn;
  assign mem_rdwrPort_w_addr = io_addr;
  assign mem_rdwrPort_w_mask = io_write;
  assign mem_rdwrPort_w_en = io_enable & (io_enable & io_write);
  assign io_dataOut = mem_rdwrPort_r_data; // @[memtest.scala 27:21 28:34]
  always @(posedge clock) begin
    if (mem_rdwrPort_w_en & mem_rdwrPort_w_mask) begin
      mem[mem_rdwrPort_w_addr] <= mem_rdwrPort_w_data; // @[memtest.scala 19:24]
    end
    mem_rdwrPort_r_en_pipe_0 <= io_enable & ~(io_enable & io_write);
    if (io_enable & ~(io_enable & io_write)) begin
      mem_rdwrPort_r_addr_pipe_0 <= io_addr;
    end
  end
// Register and memory initialization
`ifdef RANDOMIZE_GARBAGE_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
`define RANDOMIZE
`endif
`ifndef RANDOM
`define RANDOM $random
`endif
`ifdef RANDOMIZE_MEM_INIT
  integer initvar;
`endif
`ifndef SYNTHESIS
`ifdef FIRRTL_BEFORE_INITIAL
`FIRRTL_BEFORE_INITIAL
`endif
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_MEM_INIT
  _RAND_0 = {1{`RANDOM}};
  for (initvar = 0; initvar < 1024; initvar = initvar+1)
    mem[initvar] = _RAND_0[31:0];
`endif // RANDOMIZE_MEM_INIT
`ifdef RANDOMIZE_REG_INIT
  _RAND_1 = {1{`RANDOM}};
  mem_rdwrPort_r_en_pipe_0 = _RAND_1[0:0];
  _RAND_2 = {1{`RANDOM}};
  mem_rdwrPort_r_addr_pipe_0 = _RAND_2[9:0];
`endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
endmodule
`

JACKLIAO0, 2022-05-15:
I try to run a simplest demo to test whether loadMemoryFromFileInline can work correctly like following:
```
import chisel3._
import chisel3.stage.ChiselStage
import chisel3.util.experimental.loadMemoryFromFileInline

class InitMemInline(memoryFile: String = "") extends Module {
val width: Int = 32
val io = IO(new Bundle {
val enable = Input(Bool())
val write = Input(Bool())
val addr = Input(UInt(10.W))
val dataIn = Input(UInt(width.W))
val dataOut = Output(UInt(width.W))
})
val mem = SyncReadMem(1024, UInt(width.W))
// Initialize memory
if (memoryFile.trim().nonEmpty) {
loadMemoryFromFileInline(mem, memoryFile)
}
io.dataOut := DontCare
when(io.enable) {
val rdwrPort = mem(io.addr)
when (io.write) { rdwrPort := io.dataIn }
.otherwise { io.dataOut := rdwrPort }
}}

object myRtlEmitter extends App {
val prefix = "xxxx"
(new ChiselStage).emitVerilog(
new InitMemInline,
args ++ Array("-td", "./build/test"),
)
}
```

**but the generated verilog have still no readmem,like this:** 
```
module InitMemInline( input clock, input reset, input io_enable, input io_write, input [9:0] io_addr, input [31:0] io_dataIn, output [31:0] io_dataOut ); ifdef RANDOMIZE_MEM_INIT
reg [31:0] _RAND_0;
endif // RANDOMIZE_MEM_INIT ifdef RANDOMIZE_REG_INIT
reg [31:0] _RAND_1;
reg [31:0] _RAND_2;
endif // RANDOMIZE_REG_INIT reg [31:0] mem [0:1023]; // @[memtest.scala 19:24] wire mem_rdwrPort_r_en; // @[memtest.scala 19:24] wire [9:0] mem_rdwrPort_r_addr; // @[memtest.scala 19:24] wire [31:0] mem_rdwrPort_r_data; // @[memtest.scala 19:24] wire [31:0] mem_rdwrPort_w_data; // @[memtest.scala 19:24] wire [9:0] mem_rdwrPort_w_addr; // @[memtest.scala 19:24] wire mem_rdwrPort_w_mask; // @[memtest.scala 19:24] wire mem_rdwrPort_w_en; // @[memtest.scala 19:24] reg mem_rdwrPort_r_en_pipe_0; reg [9:0] mem_rdwrPort_r_addr_pipe_0; assign mem_rdwrPort_r_en = mem_rdwrPort_r_en_pipe_0; assign mem_rdwrPort_r_addr = mem_rdwrPort_r_addr_pipe_0; assign mem_rdwrPort_r_data = mem[mem_rdwrPort_r_addr]; // @[memtest.scala 19:24] assign mem_rdwrPort_w_data = io_dataIn; assign mem_rdwrPort_w_addr = io_addr; assign mem_rdwrPort_w_mask = io_write; assign mem_rdwrPort_w_en = io_enable & (io_enable & io_write); assign io_dataOut = mem_rdwrPort_r_data; // @[memtest.scala 27:21 28:34] always @(posedge clock) begin if (mem_rdwrPort_w_en & mem_rdwrPort_w_mask) begin mem[mem_rdwrPort_w_addr] <= mem_rdwrPort_w_data; // @[memtest.scala 19:24] end mem_rdwrPort_r_en_pipe_0 <= io_enable & ~(io_enable & io_write); if (io_enable & ~(io_enable & io_write)) begin mem_rdwrPort_r_addr_pipe_0 <= io_addr; end end // Register and memory initialization ifdef RANDOMIZE_GARBAGE_ASSIGN
define RANDOMIZE endif
ifdef RANDOMIZE_INVALID_ASSIGN define RANDOMIZE
endif ifdef RANDOMIZE_REG_INIT
define RANDOMIZE endif
ifdef RANDOMIZE_MEM_INIT define RANDOMIZE
endif ifndef RANDOM
define RANDOM $random endif
ifdef RANDOMIZE_MEM_INIT integer initvar; endif
ifndef SYNTHESIS ifdef FIRRTL_BEFORE_INITIAL
FIRRTL_BEFORE_INITIAL endif
initial begin
ifdef RANDOMIZE ifdef INIT_RANDOM
INIT_RANDOM endif
ifndef VERILATOR ifdef RANDOMIZE_DELAY
#RANDOMIZE_DELAY begin end else
#0.002 begin end
endif endif
ifdef RANDOMIZE_MEM_INIT _RAND_0 = {1{RANDOM}};
for (initvar = 0; initvar < 1024; initvar = initvar+1)
mem[initvar] = _RAND_0[31:0];
endif // RANDOMIZE_MEM_INIT ifdef RANDOMIZE_REG_INIT
_RAND_1 = {1{RANDOM}}; mem_rdwrPort_r_en_pipe_0 = _RAND_1[0:0]; _RAND_2 = {1{RANDOM}};
mem_rdwrPort_r_addr_pipe_0 = _RAND_2[9:0];
endif // RANDOMIZE_REG_INIT endif // RANDOMIZE
end // initial
ifdef FIRRTL_AFTER_INITIAL FIRRTL_AFTER_INITIAL
endif endif // SYNTHESIS
endmodule
```

@sequencer

sequencer, 2022-05-15:
Why @me LOL

sequencer, 2022-05-15:
1. I'm assuming you are using `chipyard` or `rocket-chip` framework, since you are using the diplomacy framework.
Those should be reported to rocket-chip or chipyard.
2. I copy-pasted your codes to https://scastie.scala-lang.org/Wmmh74xEQv2lQErzgrvicQ. You didn't give any binary to load, so the behavior of that code is correct.
I suggest you can use scastie to give an example to help the developer to debug your issue.

closed 2022-05-15 by JACKLIAO0

JACKLIAO0, 2022-05-15:
> 1. I'm assuming you are using `chipyard` or `rocket-chip` framework, since you are using the diplomacy framework.
>    Those should be reported to rocket-chip or chipyard.
> 2. I copy-pasted your codes [...]

Thanks for you reply,
  I have tried to use loadMemoryFromFileInline API with a absolute path, then readmemh occurd in the generated verilog. 

  So I closed the issue. I should open an issue in rocket-chip repository, because I still can't use loadMemoryFromFileInline API correctly in rocket-chip.