chipsalliance/chisel #2532: loadMemoryFromFileInline can not initialize memory mem of non ground type

JACKLIAO0, 2022-05-15:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: proposal

**Other information**

**Please tell us about your environment:**

    - version: 3.5.1
    - OS: `Linux IC_EDA 3.10.0-1160.53.1.el7.x86_64 #1 SMP Fri Jan 14 13:59:45 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

**What is the current behavior?**
I am using rocket-chip for my project, and I want to load a data file to a sram in rocket-chip.

I tried to use loadMemoryFromFile/loadMemoryFromFileInline to implement my needs, but during my use of it, I find some bug of it.

First, I found that firrtl parameter for sram replace 'repl-seq-mem' is conflicted with loadMemoryFromFile/loadMemoryFromFileInline, loadMemoryFromFile/loadMemoryFromFileInline can not work when using repl-seq-mem simultaneously. 

Second, When I use loadMemoryFromFile to load memory from file, it can work correctly. But when I use loadMemoryFromFileInline to load memory (non ground type), it wil occur error like this: 
```
Caused by: java.lang.RuntimeException: [module TLRAM] Cannot initialize memory mem of non ground type UInt<8>[4]
```

Is this normal? Or loadMemoryFromFileInline this API has a bug. 

Hope someone can answer my question.

thanks！

ekiwi, 2022-05-15:
> First, I found that firrtl parameter for sram replace 'repl-seq-mem' is conflicted with loadMemoryFromFile/loadMemoryFromFileInline, loadMemoryFromFile/loadMemoryFromFileInline can not work when using repl-seq-mem simultaneously.

This should ideally not be a quite failure but should throw an error or at least show you a warning. But yeah, right now it unfortunately does not.

> Is this normal? Or loadMemoryFromFileInline this API has a bug.

This is a known limitation. Memories of Vec or Bundle get translated in a way that makes it essentially impossible to support loading the content from file right now. We would like to fix that, but no one has had time to so far.

carlosedp, 2022-06-15:
As Kevin explained, if you have a memory that has for example bit masks (for example a 32bit memory width with 8bit masks), the memory is generated as "4 separated memories" and therefore the directive to import a file into that memory doesn't work (since the Verilog readmem command expect it to be a single memory instead of 4 memories).

closed 2022-08-02 by JACKLIAO0