chipsalliance/chisel #2638: type AffectsChiselPrefix is not a member of package chisel3.experimental

pxs7, 2022-07-22:
[error] ## Exception when compiling 20 sources to out\foreign-modules\rocket-chip\hardfloat\hardfloat\compile.dest\classes
[error] scala.reflect.internal.Types$TypeError: type AffectsChiselPrefix is not a member of package chisel3.experimental

AffectsChiselPrefix  is not  a member of  chisel3.experimental， why use that？

jackkoenig, 2022-07-22:
AffectsChiselPrefix **is** a member of chisel3.experimental, but currently only in Chisel 3.6-SNAPSHOT an 3.5-SNAPSHOT. It will be released as part of Chisel 3.5.4, so assuming you are using a released version of Chisel, you will need to wait for the next release to use it.

closed 2022-07-22 by jackkoenig

pxs7, 2022-07-23:
> AffectsChiselPrefix **is** a member of chisel3.experimental, but currently only in Chisel 3.6-SNAPSHOT an 3.5-SNAPSHOT. It will be released as part of Chisel 3.5.4, so assuming you are using a released version of Chisel, you will need to wait for the next release to use it.

thanks

pxs7, 2022-07-25:
> 

 when  will  Chisel 3.5.4 will be released

pxs7, 2022-07-28:
> AffectsChiselPrefix **is** a member of chisel3.experimental, but currently only in Chisel 3.6-SNAPSHOT an 3.5-SNAPSHOT. It will be released as part of Chisel 3.5.4, so assuming you are using a released version of Chisel, you will need to wait for the next release to use it.

when will Chisel 3.5.4  be released

jackkoenig, 2022-07-28:
> when will Chisel 3.5.4 be released

Soon, probably tomorrow.

jackkoenig, 2022-07-30:
I have released [Chisel v3.5.4](https://github.com/chipsalliance/chisel3/releases/tag/v3.5.4)

pxs7, 2022-07-31:
> I have released [Chisel v3.5.4](https://github.com/chipsalliance/chisel3/releases/tag/v3.5.4)

Thanks， the error stills exists, the original define for AffectsChiselPrefix as follows

    private val shouldMatchInstance: Type => Boolean = shouldMatchGen(tq"chisel3.experimental.hierarchy.Instance[_]")
    private val shouldMatchChiselPrefixed: Type => Boolean =
      shouldMatchGen(
        tq"chisel3.experimental.AffectsChiselPrefix"
      )

it seams that mill can not recognise the code, and fails found AffectsChiselPrefix,  [mill version =0.10.4],

pxs7, 2022-07-31:
> I have released [Chisel v3.5.4](https://github.com/chipsalliance/chisel3/releases/tag/v3.5.4)

compile success，  thanks