chipsalliance/chisel #2660: PROBLEM IN FIXEDPOINT DIVISION

BhattSoham, 2022-08-04:
When I am running a code for testing in fixedpoint, I am getting an error. Can anyone help me with the code???
Code: 
package mypack
import chisel3._
import chisel3.util._
import chisel3.experimental.FixedPoint
import chisel3.internal.ChiselException
//import chisel3.internal.firrtl.{BinaryPoint, KnownBinaryPoint}
//import chisel3.internal.sourceinfo.{SourceInfo, SourceInfoTransform}
import scala.language.experimental.macros
//import Node._
//import ChiselError._
//import hardfloat._

 
class FPMul extends Module {

 val io = IO(new Bundle {
  val a = Input(FixedPoint(4.W, 2.BP))
  val b = Input(FixedPoint(4.W, 2.BP))
  val p = Output(FixedPoint(8.W, 2.BP))
})

      
io.p := io.a + io.b / 6.0.F(0.BP)

}
object FPMul extends App {
 (new chisel3.stage.ChiselStage).emitVerilog(new FPMul())
}

Error:
FPMul
[info] - should pass *** FAILED ***
[info**]   **chisel3.internal.ChiselException: division is illegal on FixedPoint types****
[info]   at ... ()
[info]   at mypack.FPMul.$anonfun$new$1(FPMul.scala:26)
[info]   at chisel3.Data.$anonfun$$colon$eq$1(Data.scala:748)
[info]   at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
[info]   at chisel3.internal.prefix$.apply(prefix.scala:31)
[info]   at chisel3.Data.$colon$eq(Data.scala:748)
[info]   at mypack.FPMul.<init>(FPMul.scala:26)
[info]   at mypack.FPMulTest.$anonfun$new$2(FPMulTest.scala:14)
[info]   at ... ()
[info]   at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)

Your help is highly appreciated.

Martoni, 2022-08-05:
As I can understand,[ it's not been implemented ](https://github.com/chipsalliance/chisel3/blob/v3.5.3/core/src/main/scala/chisel3/Bits.scala#L1495): 

```Scala
    override def do_/(that: FixedPoint)(implicit sourceInfo: SourceInfo, compileOptions: CompileOptions): FixedPoint =
      throwException(s"division is illegal on FixedPoint types")
```

BhattSoham, 2022-08-05:
While using this, it is throwing this:
not found: value throwException
[error]       throwException(s"division is illegal on FixedPoint types")

jackkoenig, 2022-08-09:
The error is telling you that you cannot do division with fixed point types. If you want to do division, you will need to implement it yourself.

closed 2022-08-09 by jackkoenig