chipsalliance/chisel #2681: FIXED POINT LOOP QUERY

BhattSoham, 2022-08-14:
Hi there,
   I am new to chisel. I just need to know that I am performing an addition of a fixed point. But whenever I am performing any loop operation, it is giving the same result again and again. Can anyone help me fixing the code:
   
Code: 
import chisel3._
import chisel3.util._
import chisel3.experimental.FixedPoint

   class Add extends Module {
 val io = IO(new Bundle {
  val x_in = Input(FixedPoint(32.W, 8.BP))
  val y_in = Input(FixedPoint(32.W, 8.BP))
  val h = Input(FixedPoint(32.W, 8.BP))
  val x_out = Output(FixedPoint(64.W, 16.BP))
  //val y_out = Output(FixedPoint(64.W, 16.BP))
})
 
io.x_out := io.x_in + io.h
 
}

TestBench:

import chisel3._
import chisel3.util._
import chisel3.experimental.FixedPoint
import chisel3.tester._

class AddTest extends AnyFlatSpec with Matchers {
"Add" should "pass" in {
test(new Add) { c => 
for(i<- 0 until 5) {
 c.io.x_in.poke(15.F(8.BP))
 

 c.io.h.poke(0.1.F(8.BP))

 c.clock.step(1)

 println( "Resultx:" + c.io.x_out.peek() )
} 

Result:
Resultx:FixedPoint<21><<16>>(15.1015625)
Resultx:FixedPoint<21><<16>>(15.1015625)
Resultx:FixedPoint<21><<16>>(15.1015625)
Resultx:FixedPoint<21><<16>>(15.1015625)
Resultx:FixedPoint<21><<16>>(15.1015625)

Your help is appreciated.

ekiwi, 2022-08-14:
Same thing as in your previous question. If you give the circuit the same values, the result will be the same.

Try applying different values and you will see a different result:

```
c.io.x_in.poke(15.F(8.BP))
c.io.h.poke(0.1.F(8.BP))
println( "Resultx:" + c.io.x_out.peek())
c.clock.step(1)
c.io.x_in.poke(3.F(8.BP))
c.io.h.poke(4.F(8.BP))
println( "Resultx:" + c.io.x_out.peek())
c.clock.step(1)
```

closed 2022-08-14 by ekiwi

BhattSoham, 2022-08-14:
No actually I want to know if like h is constant which is 0.1 and I want to increment the value if x_in by 0.1 in every step till 5 in the loop. Will that be done?

ekiwi, 2022-08-14:
> Suppose the first one is coming as 15.10... The second one will be 15.2. I want that using the loop function addition. Any clue if it can be done. I don't want to add manually.

I would recommend having a look at a Scala tutorial. What you are asking for is not really Chisel specific. If you make a `var` in Scala, you can increment it, just like any variable in Java. Something like:

```.scala
var a = BigDecimal("15.1")
println(a) // prints: 15.1
a += 0.1
println(a) // prints: 15.2
```