chipsalliance/chisel #2771: Recursive RegNext Breaks Naming

seldridge, 2022-10-12:
I preface this by I think this is low priority and relatively rare to see.

Consider the following circuit that describe a register that increments by one, from a reset value of zero, forever:

```scala
class NonRecursiveRegister extends Module {

  val a = RegInit(0.U(8.W))
  a := a + 1.U
  
}
```

Now consider this circuit rewritten using a `RegNext`:

```scala
class RecursiveRegister extends Module {
  val a: UInt = RegNext(aNext, 0.U(8.W))
  val aNext = a + 1.U
}
```

This then crashes without a great error:

```
java.lang.NullPointerException: Cannot invoke "chisel3.Data.cloneTypeFull()" because "<local7>" is null
	at ... ()
	at Playground$RecursiveRegister$$anonfun$3$$anonfun$apply$2.apply(main.scala:15)
	at Playground$RecursiveRegister$$anonfun$3$$anonfun$apply$2.apply(main.scala:15)
	at chisel3.internal.prefix$.apply(prefix.scala:48)
	at Playground$RecursiveRegister$$anonfun$3.apply(main.scala:15)
	at Playground$RecursiveRegister$$anonfun$3.apply(main.scala)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at Playground$RecursiveRegister.<init>(main.scala:15)
	at Playground$$anonfun$5.apply(main.scala:19)
	at Playground$$anonfun$5.apply(main.scala:19)
	at ... ()
```

I can recognize disallowing this with a better error message. However, this seems like something that should be capable of being expressed. I assume that it just needs a tweak to name prefixing to detect when this is happening and either bail (generate a bad name) or there is a way to still generate a good name.

Scastie link: https://scastie.scala-lang.org/seldridge/K8KGN2kORfqPudn2dk2qJA/46

aswaterman, 2022-10-12:
I agree this should be allowed, with good naming preferable to poor naming, but with poor naming preferable to an error (no matter how good the message).

aswaterman, 2022-10-12:
I misread the original code so deleted my earlier response. I view the functionality as nice to have, but not strictly necessary, and so I would be curious to see what a functionally correct resolution would look like.

Failing that, a better error message would be nice, but it falls into the broader class of forward references to class members not yet being initialized.

seldridge, 2022-10-12:
FWIW, this has been a `NullPointerException` going all the way back to Chisel 3.0.0. I'm fine with just a better message here. 😅 

I got to this point trying to write a pithy example involving `val a: UInt = RegNext(a + 1.U, 0.U(4.W))`. This seemed like it should work as the register next value is disconnected from the register declaration at the FIRRTL level (which is admittedly many layers away from Chisel). However, naming and (later) the type of the register derives from the next value. Hence, this is legal Scala, but not legal Chisel.

It's possible that closer coupling between Chisel and Scala would enable this, so long as the forward type annotation was present.