chipsalliance/chisel #2773: MixedVec and CompatibilityMode don't work consistently together

mwachs5, 2022-10-13:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

3.5.4 Scastie demonstrating the issue:

https://scastie.scala-lang.org/oAJddondSVWO1XcqVNMNag

```
object Interrupts {
    import chisel3.{Bool, Output, Vec}

    def bundle() = Vec(3, Bool())
  }

object OldStuff {

    import Chisel._

    import chisel3.util.MixedVec

    class LazyModuleImp[T <: Data](bundleGen: () => Seq[T]) extends Module {
     val io = IO(new Bundle {
        val out: MixedVec[T] = MixedVec(bundleGen())
        val in:  MixedVec[T] = Flipped(MixedVec(bundleGen()))
      })
      
      io.out := RegNext(io.in)
    }

    class OldInterrupts extends LazyModuleImp(bundleGen = () => Seq.fill(3) { Interrupts.bundle() })
  }
  object NewStuff {
    import chisel3._
    import chisel3.experimental.hierarchy.Instance

    class NewInterrupts extends OldStuff.LazyModuleImp(bundleGen = () => Seq.fill(3) { Interrupts.bundle() })

    class Example extends Module {
      val oldMod = Module(new OldStuff.OldInterrupts)
      val newMod = Module(new NewInterrupts)

      // Both of these don't work
      //newMod.io.in <> DontCare
      oldMod.io.in <> DontCare
    }
  }
  // This is fine
 (new chisel3.stage.ChiselStage).emitVerilog(new OldStuff.OldInterrupts, Array("--full-stacktrace"))
  // This does not work
 (new chisel3.stage.ChiselStage).emitVerilog(new NewStuff.Example, Array("--full-stacktrace"))
```

**What is the current behavior?**

If you put a MixedVec inside the IO of a Chisel.Module without specifying their direction it will *think* that output direction is inferred for the leaf elements, when in fact it isn't. The IO should at least complain if it wants to enforce directionality there, but it doesn't, and then you get a confusing message later when you try to connect to that in a chisel3.<>.

**What is the expected behavior?**

You should be able to <> to Chisel3 IOs that has had to infer direction within a chisel3 module without error.

**Please tell us about your environment:**
    - version: `3.5.4` --note this issue does NOT exist on master because we decided to consistently infer Output direction but that change is not on 3.5.x branch.

**Other Information**

**What is the use case for changing the behavior?**

Being able to use MixedVecs in Chisel code mixed with chisel3 modules in a consistent way (notably this is important for rocket-chip diplomacy use cases where mixing of Chisel and chisel3 is happening).

Suggested fix is to pass compileOptions through MixedVec class and its companion object methods, a fix which I have tested and gives expected behavior

mwachs5, 2022-10-19:
Or, assuming we just don't expect MixedVec to work this way (which is fine), then we need to be consistent and error if you try to do the above while you are building the IO. I think the right fix is just to get rid of the "nasty compatibility mode shim" in Aggregate.scala, because I don't understand the use case for this -- by the time this is called, all Unspecified directions should have been resolved with the assignCompat thing, so we shouldn't have special handling for this case (again this code is already gone from master this is just about 3.5.x)

jackkoenig, 2024-05-21:
Compatibility mode has been removed.

closed 2024-05-21 by jackkoenig