chipsalliance/chisel #2808: java.nio.file.NoSuchFileException

pxs7, 2022-10-25:
src 

  val mfile = "./data.txt"
  when(lReset) {
     loadMemoryFromFile(memory, mfile)
  }

error:
  - should pass *** FAILED ***
  java.nio.file.NoSuchFileException: data.txt
  at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
  at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:106)
  at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
  at java.base/sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:218)
  at java.base/java.nio.file.Files.newByteChannel(Files.java:380)
  at java.base/java.nio.file.Files.newByteChannel(Files.java:432)
  at java.base/java.nio.file.spi.FileSystemProvider.newInputStream(FileSystemProvider.java:422)
  at java.base/java.nio.file.Files.newInputStream(Files.java:160)
  at os.Path.getInputStream(Path.scala:491)
  at os.read$lines$stream$$anon$3.generate(ReadWriteOps.scala:345)
  at geny.Generator.foreach(Generator.scala:49)
  at geny.Generator.foreach$(Generator.scala:49)
  at os.read$lines$stream$$anon$3.foreach(ReadWriteOps.scala:343)
  at geny.Generator.toBuffer(Generator.scala:127)
  at geny.Generator.toBuffer$(Generator.scala:125)
  at os.read$lines$stream$$anon$3.toBuffer(ReadWriteOps.scala:343)
  at geny.Generator.toArray(Generator.scala:131)
  at geny.Generator.toArray$(Generator.scala:131)
  at os.read$lines$stream$$anon$3.toArray(ReadWriteOps.scala:343)
  at os.read$lines$.apply(ReadWriteOps.scala:330)
  at firrtl.FileUtils$.getLines(FileUtils.scala:118)
  at firrtl.FileUtils$.getLines(FileUtils.scala:112)
  at treadle.executable.MemoryInitializer.doInitialize(Memory.scala:669)
  at treadle.executable.MemoryInitializer.$anonfun$initializeMemoriesFromFiles$1(Memory.scala:685)
  at treadle.executable.MemoryInitializer.$anonfun$initializeMemoriesFromFiles$1$adapted(Memory.scala:684)      
  at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
  at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
  at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
  at treadle.executable.MemoryInitializer.initializeMemoriesFromFiles(Memory.scala:684)
  at treadle.executable.ExecutionEngine$.apply(ExecutionEngine.scala:681)
  at treadle.TreadleTester.<init>(TreadleTester.scala:43)
  at treadle.stage.phases.CreateTester$.transform(CreateTester.scala:19)
  at treadle.stage.phases.CreateTester$.transform(CreateTester.scala:10)
  at treadle.stage.TreadleTesterPhase.$anonfun$transform$1(TreadleTesterPhase.scala:21)
  at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
  at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
  at scala.collection.immutable.List.foldLeft(List.scala:91)
  at treadle.stage.TreadleTesterPhase.transform(TreadleTesterPhase.scala:21)
  at chiseltest.simulator.TreadleSimulator$.createContext(TreadleSimulator.scala:38)
  at chiseltest.internal.BackendExecutive$.start(BackendExecutive.scala:36)
  at chiseltest.defaults.package$.createDefaultTester(defaults.scala:21)
  at chiseltest.ChiselScalatestTester$TestBuilder.apply(ChiselScalatestTester.scala:31)
  at core.CoreTester.simpletest(Test.scala:26)
  at core.CoreTester.$anonfun$new$1(Test.scala:21)
  at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
  at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
  at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
  at org.scalatest.Transformer.apply(Transformer.scala:22)
  at org.scalatest.Transformer.apply(Transformer.scala:20)
  at org.scalatest.flatspec.AnyFlatSpecLike$$anon$5.apply(AnyFlatSpecLike.scala:1812)
  at org.scalatest.TestSuite.withFixture(TestSuite.scala:196)
  at org.scalatest.TestSuite.withFixture$(TestSuite.scala:195)
  at core.CoreTester.chiseltest$ChiselScalatestTester$$super$withFixture(Test.scala:17)
  at chiseltest.ChiselScalatestTester.$anonfun$withFixture$1(ChiselScalatestTester.scala:82)

 loadMemoryFromFile seam cannot find file which exits

pxs7, 2022-10-25:
i use linux cmd “find” in scala src code，  it seem that java  can just find  jar file

ekiwi, 2022-10-25:
Can you show us your directory layout? In general the path should be relative to the directory that contains your `build.sbt`

ekiwi, 2022-10-25:
> i use java bloopCli directly build the src，not sbt. i hase used mill build. it work. But bloop can not

yeah, sorry, no idea how bloop works

Try printing out the current working directory from the test and see what it is.

closed 2022-10-25 by ekiwi

pxs7, 2022-10-26:
> > i use java bloopCli directly build the src，not sbt. i hase used mill build. it work. But bloop can not
> 
> yeah, sorry, no idea how bloop works
> 
> Try printing out the current working directory from the test and see what it is.
do you mean
/mnt/c/workspace/chiseltest ?

pxs7, 2022-10-26:
> > i use java bloopCli directly build the src，not sbt. i hase used mill build. it work. But bloop can not
> 
> yeah, sorry, no idea how bloop works
> 
> Try printing out the current working directory from the test and see what it is.

/mnt/c/workspace/chiseltest,  i use wsl2 in windows

ekiwi, 2022-10-26:
> /mnt/c/workspace/chiseltest

Is your file that you are trying to load in that same directory?

pxs7, 2022-10-27:
> > /mnt/c/workspace/chiseltest
> 
> Is your file that you are trying to load in that same directory?

yes,  absolute path 

/mnt/c/workspace/chiseltest/data.txt