chipsalliance/chisel #2816: Inconsistent width handling in Interval creation

ivanvig, 2022-10-30:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

Open a scala console

**What is the current behavior?**

When I try to create a literal Interval from a decimal in the following way:

```
scala> 7.0.I(3.W, 2.BP)
```

I get an error:

```
chisel3.internal.ChiselException: Error literal interval value 28 is too many bits for specified width 3
  at chisel3.experimental.Interval$.Lit(Bits.scala:2474)
  at chisel3.experimental.Interval$Implicits$fromBigDecimalToLiteralInterval.I(Bits.scala:2581)
  at $anonfun$res49$2(<console>:1)
  at chisel3.internal.prefix$.apply(prefix.scala:48)
  at $anonfun$res49$1(<console>:1)
  at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
  ... 59 elided
```

Which is what I would expect, since 3bits is not enough to represent the decimal value, on the other hand, using the Interval object method:

```
scala> Interval.fromDouble(7.0, width=3.W, binaryPoint=2.BP)
val res50: chisel3.experimental.Interval = Interval<6>(28)
```

doesn't fail, but instead ignores the width value and sets a new valid one, which I think could introduce unexpected and difficult to debug bugs.

Another issue that I found is when creating a new Interval using:

```
scala> Interval(3.W, 2.BP)
val res51: chisel3.experimental.Interval = Interval<5>
```

It creates an Interval with a width of 5bits, while I would've expected a width of 3bits.

**Please tell us about your environment:**

chisel version: 3.5.3
OS: Linux 5.19.13-arch1-1

ivanvig, 2022-10-30:
I think I fixed both bugs, I can create a pr for both fixes or maybe a pr per fix. Please tell me how you would prefer it

mwachs5, 2023-06-06:
since Intervals are no longer supported, I think we should close out this issue.

closed 2023-06-06 by mwachs5