chipsalliance/chisel #2821: Error with Huge array requirement

hanm2019, 2022-11-01:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

I get the Error `java.lang.OutOfMemoryError: Required array length 2147483640 + 18 is too large` when I try to generate a gemmini model with 64x64 PEs, it seems that the chisel3 gets error during the firrtl stage. However, the JAVA VM sets the array's size to be smaller than 2^32 -1 , so how can I generate the RTL with a large parameter?

**Please tell us about your environment:**
version: `3.5.2`
OS: ubuntu 18.04 with 256GB memory.
chipyard version: 1.8.1

```
  Output clock buildtop_reference_clock, requested: 100.0 MHz, actual: 100.0 MHz (division of 1)
[error] (run-main-0) java.lang.OutOfMemoryError: Required array length 2147483640 + 18 is too large
[error] java.lang.OutOfMemoryError: Required array length 2147483640 + 18 is too large
[error]         at java.base/jdk.internal.util.ArraysSupport.hugeLength(ArraysSupport.java:649)
[error]         at java.base/jdk.internal.util.ArraysSupport.newLength(ArraysSupport.java:642)
[error]         at java.base/java.lang.AbstractStringBuilder.newCapacity(AbstractStringBuilder.java:257)
[error]         at java.base/java.lang.AbstractStringBuilder.ensureCapacityInternal(AbstractStringBuilder.java:229)
[error]         at java.base/java.lang.AbstractStringBuilder.append(AbstractStringBuilder.java:582)
[error]         at java.base/java.lang.StringBuilder.append(StringBuilder.java:179)
[error]         at scala.collection.mutable.StringBuilder.$plus$plus$eq(StringBuilder.scala:183)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:72)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:299)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:74)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:96)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:110)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:246)
[error]         at firrtl.ir.Serializer$.$anonfun$s$9(Serializer.scala:260)
[error]         at firrtl.ir.Serializer$.$anonfun$s$9$adapted(Serializer.scala:260)
[error]         at scala.collection.immutable.Stream.foreach(Stream.scala:533)
[error]         at firrtl.ir.Serializer$.s(Serializer.scala:260)
[error]         at firrtl.ir.Serializer$.serialize(Serializer.scala:36)
[error]         at firrtl.ir.Serializer$.serialize(Serializer.scala:17)
[error]         at chisel3.internal.firrtl.Emitter$.emit(Emitter.scala:11)
[error]         at chisel3.stage.phases.Emitter.$anonfun$transform$1(Emitter.scala:45)
[error]         at scala.collection.TraversableLike.$anonfun$flatMap$1(TraversableLike.scala:245)
[error]         at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error]         at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[error]         at scala.collection.TraversableLike.flatMap(TraversableLike.scala:245)
[error]         at scala.collection.TraversableLike.flatMap$(TraversableLike.scala:242)
[error]         at scala.collection.AbstractTraversable.flatMap(Traversable.scala:108)
[error]         at chisel3.stage.phases.Emitter.transform(Emitter.scala:42)
[error]         at chisel3.stage.phases.Emitter.transform(Emitter.scala:25)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:248)
[error]         at firrtl.options.Translator.transform$(Phase.scala:248)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.DependencyManager.$anonfun$transform$5(DependencyManager.scala:280)
[error]         at firrtl.Utils$.time(Utils.scala:181)
[error]         at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:280)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
[error]         at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
[error]         at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
[error]         at chisel3.stage.ChiselStage.run(ChiselStage.scala:45)
[error]         at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error]         at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:248)
[error]         at firrtl.options.Translator.transform$(Phase.scala:248)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[... 24 lines cut ...]
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:248)
[error]         at firrtl.options.Translator.transform$(Phase.scala:248)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error]         at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error]         at logger.Logger$.makeScope(Logger.scala:135)
[error]         at firrtl.options.Stage.transform(Stage.scala:47)
[error]         at firrtl.options.Stage.execute(Stage.scala:58)
[error]         at firrtl.options.StageMain.main(Stage.scala:71)
[error]         at chipyard.Generator.main(Generator.scala)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
[error]         at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.base/java.lang.reflect.Method.invoke(Method.java:568)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
```

jackkoenig, 2022-11-05:
The issue is that there is a single module that is emitting to over 2 GiB of FIRRTL text. This is fixed on master and 3.5.x, the fix will be released in 3.5.5 (which we really need to just go ahead and release).

I will warn that even with 3.5.5 when it's released, you will probably run into problems in the Scala FIRRTL Compiler. It might be worth trying to reduce the maximum module size in Gemmini to have an overall more scalable result. You might also consider looking into the MLIR FIRRTL Compiler (MFC) (see https://github.com/sifive/chisel-circt).

hanm2019, 2022-11-06:
I try to use BlackBox to solve this problem. In fact, I generate a Verilog file with 128x128 PEs only, and the make the Mesh model as a BlackBox to avoid the huge array. However, I met another problem: `Method too large`, I guess this is because the Mesh with 128x128 PEs have more than 1000+ io signals?

ekiwi, 2022-11-07:
> I will warn that even with 3.5.5 when it's released, you will probably run into problems in the Scala FIRRTL Compiler.

Wouldn't using Definition/Instance for this also solve the issue? I assume that all PEs are essentially the same module.