chipsalliance/chisel #2848: Cause: chisel3.internal.ChiselException: Error: attempted to instantiate a Module without wrapping it in Module()

pxs7, 2022-11-15:
when i just want to use a function of module (test), 
   when use Module(new test), a verilog module which is not expected  is generated, 
   when use  new test,  the  following error  
      "Cause: chisel3.internal.ChiselException: Error: attempted to instantiate a Module without wrapping it in Module()"
   comes out
  
any way that can help me use the method of test, but no extra verilog module is generated

jackkoenig, 2022-11-22:
Can you provide more information about what you are doing? Ideally with some code examples?

pxs7, 2022-11-22:
> Can you provide more information about what you are doing? Ideally with some code examples?

some sub module config information is put into in submodule，before construct top module，  such submodule information is needed  through  sub module information  to determine  full config information

pxs7, 2022-11-22:
> Can you provide more information about what you are doing? Ideally with some code examples?

some sub module config information is put into in submodule，before construct top module， such submodule information is needed through sub module function to determine full config information

jackkoenig, 2022-11-22:
This isn't enough information to help debug the problem. Please try to minimize the problem with a small example or other code sketch. Ideally, you could show the bug in a Scastie: https://scastie.scala-lang.org/9AZWsMbFTpWPfXppd0W8VQ