chipsalliance/chisel #2879: SyncReadMem: read(addr) problem inside ShiftRegisterMem

milovanovic, 2022-12-09:
I believe that the [commit](https://github.com/chipsalliance/chisel3/commit/29665743acff120bc87ee997890d7f952317144e) broke some edge cases of [`ShiftRegisterMem`](https://github.com/ucb-bar/rocket-dsp-utils/blob/master/src/main/scala/craft/ShiftRegisterMem.scala). A detailed demonstration of this bug can be found [here](https://github.com/milovanovic/ShiftRegisterMemDemo).

jerryz123, 2022-12-21:
I have not looked too closely, but I am inclined to believe that this is actually a bug in the implementation of ShiftRegisterMem. It should not rely on the undefined read-data of a SyncReadMem when the read-enable is low.

milovanovic, 2022-12-21:
You were right. The problem was indeed inside the `ShiftRegisterMem`. It should be fixed now, so this issue can be closed. Thanks.

closed 2022-12-31 by sequencer