chipsalliance/chisel #2885: DontCare in Mux should remove mux logic

dbear496, 2022-12-12:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
Sometimes I have a DontCare show up in a mux like so:
```
m := Mux(sel, a, DontCare)
```
Currently, this emits Verilog like this:
```
assign m = sel ? a : 8'h0;
```
However, the evaluation of `sel` could have been optimized out, and that is no longer possible with this Verilog. Since the third argument of the mux is a DontCare, and not actually a 0, this could have been written:
```
assign m = a;
```
This optimizes out the evaluation of `sel` and the mux, and it does not force `m` to 0 when it doesn't have to.

**Describe the solution you'd like**
When one of the data arguments of a Mux is DontCare, the expression should trivialize to the other data argument.
```
Mux(sel, a, DontCare) -> a
Mux(sel, DontCare, b) -> b
Mux(sel, DontCare, DontCare) -> DontCare
```

**Additional context**
Here is a demonstration in Scastie: https://scastie.scala-lang.org/GVHwYBWOS9ek14Y07cXPLA. Of course this is simplified for the sake of demonstration; in reality, the DontCare may appear through some more obscure means.

**What is the use case for implementing this feature?**
This can improve the performance of generated Verilog by removing some unnecessary logic when dealing with DontCare.

edwardcwang, 2022-12-12:
`Mux(DontCare, a, b) -> ?`

aswaterman, 2022-12-12:
Mux(DontCare, a,b) can become a or b. It shouldn’t become DontCare, because that doesn’t do the right thing for the case that a === b (when the result is well defined regardless of the condition).

seldridge, 2022-12-12:
While I fully agree that this is a legal optimization, it has historically never been something that the Scala-based FIRRTL Compiler (SFC) has done. Specifically, the SFC has a complicated, context-dependent interpretation of invalid that is [documented here](https://circt.llvm.org/docs/Dialects/FIRRTL/RationaleFIRRTL/#interpretation-of-undefined-behavior). Very concretely, because the SFC never did this optimization, users may inadvertently rely on it and changing this may break user code.

(We tried to turn this on and it broke a bunch of code. 🥲)

Naturally, this isn't where we want to be indefinitely!

The FIRRTL spec now has a more clear definition of what invalid is (which is the FIRRTL construct that is created for Chisel's `DontCare`). This has an ["indeterminate value"](https://github.com/chipsalliance/firrtl-spec/blob/main/spec.md#indeterminate-values). Where we want to go with this is to look at enabling optimizations that will more aggressively choose _some value_ for a _specific_ `DontCare` based on its uses. 

For the question of `Mux(Invalid, a, b) -> ?` this would then open the possibility for a FIRRTL compiler to choose some value for that specific invalid (which is the same thing as choosing either `a` or `b` to replace the `Mux` with). Current SFC semantics mandate that that `Invalid` is zero which potentially produces a sub-optimal result.

All of this is complicated right now as we complete the transition from the SFC to the MLIR-based FIRRTL Compiler (MFC). While we still having two FIRRTL compilers, making these types of changes isn't feasible due to the SFC missing these types of optimizations.

dbear496, 2022-12-13:
Thanks for the detailed response. It's good to hear this is on the agenda even though it is not feasible at the moment.