chipsalliance/chisel #2899: Error when using an Experimental Features(loadMemoryFromFileInline)

mingerfan, 2022-12-19:
**Type of issue**: Bug Report
**Please provide the steps to reproduce the problem**:
I have met a problem when I am trying an Experimental Features(loadMemoryFromFileInline). 
I plan to construct a memory class with same io but different initial file(as initialization input) in order to use it as a template to create a new ROM conveniently
```scala
class MemRom(addr_width: Int, out_width: Int, init_file: String) extends AbsRom(addr_width, out_width) {
    val mem = SyncReadMem(max_num, UInt(out_width.W))
    loadMemoryFromFileInline(mem, init_file)

    io.out := mem.read(io.addr)
}
```
Set the same width
```scala
val fangMem = Module(new MemRom(8, 8, "fangbo.hex"))
val triMem = Module(new MemRom(8, 8, "sanjiaobo.hex"))
```

**What is the current behavior?**
it casts error.
```shell
[error] (run-main-0) firrtl.FirrtlUserException: At least one dedupable annotation did not deduplicate: got non-local annotation MemoryFileInlineAnnotation(~WaveSigBox|WaveSigBox/triMem:MemRom>mem,sanjiaobo.hex,Hex) from [[DedupAnnotationsTransform]]
[error] firrtl.FirrtlUserException: At least one dedupable annotation did not deduplicate: got non-local annotation MemoryFileInlineAnnotation(~WaveSigBox|WaveSigBox/triMem:MemRom>mem,sanjiaobo.hex,Hex) from [[DedupAnnotationsTransform]]
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
```
**What is the expected behavior?**
I have thought chisel would recognize them as two different modules, but it seems it didn't.
**Please tell us about your environment:**
    - version: `3.5.4`

**What is the use case for changing the behavior?**
When I set different width, It works and generates verilog
```scala
   val fangMem = Module(new MemRom(8, 8, "fangbo.hex"))
   val triMem = Module(new MemRom(8, 9, "sanjiaobo.hex"))
```

```verilog
  MemRom fangMem ( // @[WaveSigBox.scala 13:25]
    .clock(fangMem_clock),
    .io_addr(fangMem_io_addr),
    .io_out(fangMem_io_out)
  );
  MemRom_1 triMem ( // @[WaveSigBox.scala 14:24]
    .clock(triMem_clock),
    .io_addr(triMem_io_addr),
    .io_out(triMem_io_out)
  );
```

carlosedp, 2022-12-19:
Currently you can do something like this in your memory module:

```scala
  // This prevents deduping this memory module
  // Ref. https://github.com/chipsalliance/firrtl/issues/2168
  val dedupBlock = WireInit(mem.hashCode.U)
```

As a workaround to prevent dedup. There's a ref issue on firrtl repo I opened some time ago.

Full example here: https://github.com/carlosedp/chisel-playground/blob/master/blockram/src/main/scala/utils/MultiMemory.scala

mingerfan, 2022-12-21:
Solved it, thanks carlosedp. I will close this issue.

closed 2022-12-21 by mingerfan