chipsalliance/chisel #2908: Hi, is chisel language supporting scala3 around the corner?

TurnOffNOD, 2022-12-22:
**Type of issue**: scala 3 support.

**Is your feature request related to a problem? Please describe.**
Now scala is developing new version scala 3, and scala users are much lesser than python, so changing from scala 2 to scala 3 will be much faster than python 2 changing to python3.

Maybe it is better to support scala 3 faster.

**Additional context**
SpinalHDL is preparing to support scala 3 from version 1.6.0: https://github.com/SpinalHDL/SpinalHDL/releases/tag/v1.6.0

Dolu1990, 2023-05-27:
So i implemented Scala 3 support for SpinalHDL.
- SpinalHDL doesn't use macro, so didn't had issue on that part
- Update of the Scala compiler plugin wasn't too hard
- So far, seems to me that Scala 3 lost some of the internal DSL vibe compared to Scala 2 (from the our use case perspective)
- Anonymous class realy aren't great in Scala 3, conflicting with io Bundle and much more with Area (https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/area.html?highlight=area#area)
- Did hit quite a few Scala compiler crash (real crash with stack trace)
- Seq being now immutable by default change is a pain XD (if i remember well)
- SpinalHDL lib, which is a large code base using SpinalHDL core, scream Scala compiler crashes / errors ^^

So overall, i'm not realy happy about how Scala 3 turned, from my perspective / usecase, it is kinda like python2 -> python3.
Seems like more paine and suffering than anything else, not event considering the end user side of things XD

mwachs5, 2023-06-15:
@Dolu1990 could you elaborate a bit on "So far, seems to me that Scala 3 lost some of the internal DSL vibe compared to Scala 2 (from the our use case perspective)"? What are some examples of things that no longer work well?

jackkoenig, 2023-06-15:
Hey Charles, thanks for letting us know about your experience.

Per the Anonymous classes, we actually provided early feedback that led to them adding a [reflective form of Selectable](https://docs.scala-lang.org/scala3/guides/migration/incompat-type-inference.html#reflective-type), so I think anonymous types can be okay:
```scala
abstract class Bundle extends reflect.Selectable

val foo = new Bundle {
  val x = 3
  val y = "hi"
}

foo.x // 3: scala.Int
```
(Scala 3.3 Scastie: https://scastie.scala-lang.org/ElSvOOAATAeC4hyBGpqmeg)

I very much would like to upgrade to Scala 3, I'm glad to hear that upgrading your compiler plugin wasn't too bad. Many of our macros go away (chained apply in particular is made obsolete by `using`), SourceInfo should be easy, but `@instantiable` and `@public` may be tricky.

The move away from DSL is somewhat unfortunate but has seemed to be the zeitgeist in Scala for a few years now 😑. I think it's mostly okay from the Chisel perspective but I should check more closely.

Seq being immutable was a Scala 2.13 change and yeah that was a pretty big pain, no lie 😅.

Compiler crashes aren't great, I'm hopeful that we're so late that many have been resolved by now but we shall see!

Dolu1990, 2023-06-16:
Hi,

> could you elaborate a bit on "So far, seems to me that Scala 3 lost some of the internal DSL vibe compared to Scala 2 (from the our use case perspective)"? What are some examples of things that no longer work well?

@mwachs5 So mostly 3 things i can remember which gived me that feeling : 
- More restrictions in the way the scala functions signature will conflict with each others (def f  def f() val f)
- Can't access field of anonymous classes without Selectable (with crashes  / bug)
- Looking at the Scala 3 dev threads, the "right" way to do things was put forward, from a narrow perspective (limiting the scope of possibilities things), instead of the freedom of the user to make his own choices

@jackkoenig Hi ^^

> Per the Anonymous classes, we actually provided early feedback that led to them adding a [reflective form of Selectable](https://docs.scala-lang.org/scala3/guides/migration/incompat-type-inference.html#reflective-type), so I think anonymous types can be okay:

Yes it is what i used too, for basics it works well, but in more imbricated cases, the Scala 3 compiler had failures : 
- Not providing access to the field anymore (at compilation)
- Compiler crashes

Note, those were related to having anon Selectable in anon Selectable and so one.

> The move away from DSL is somewhat unfortunate but has seemed to be the zeitgeist in Scala for a few years now 

:(

> I think it's mostly okay from the Chisel perspective but I should check more closely.

Main issue from a user perspective, is that, as hardware is subject to much more testing / nasty failure modes than software.
People will hate modifying a working code base, as it would be a risk + time require to tests everything again. (also, very often, testing isn't fully automated, ex FPGA emulation and so on)

> Seq being immutable was a Scala 2.13 change and yeah that was a pretty big pain, no lie sweat_smile.

Yes XD

> Compiler crashes aren't great, I'm hopeful that we're so late that many have been resolved by now but we shall see!

That was my hope too :D, so i made all the work on scala 3   14 months ago. I think the main reason of the crashes were related to the following Area feature  :

```scala
val x new Area{
  val y = new Area{
    val x = new Area{
      val miaou = Reg(UInt(8 bits))
    }
  }
}
x.y.z.miaou := 0
```

You may not hit those.

Since then, scala 3 migration is on hold.

edwardcwang, 2024-07-19:
Newer duplicate here #3967

closed 2024-07-19 by edwardcwang