chipsalliance/chisel #2920: suggestName does not work for IO ports

zhutmost, 2023-01-05:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

I want to rename the IO port `io.aaa` of Module A with `.suggestName("ccc")`, but nothing happened in the generated RTL code. The port name is still "aaa".

The following code can be reproduced in Scastie with Chisel 3.6.0-M1.

```scala
import chisel3._

case class A() extends Module {
  val io = IO(new Bundle {
    val aaa = Input(Bool())
    val bbb = Output(Bool())
  })

  io.bbb := io.aaa
}

class FinalA extends A() {
  io.aaa.suggestName("ccc")
}

class Top extends Module {
  val io = IO(new Bundle {
    val a = Input(Bool())
    val b = Output(Bool())
  })

  val uMod = Module(new FinalA)
  uMod.io.aaa := io.a
  io.b := uMod.io.bbb
}

val pretty = Array(
  "--emission-options", "disableMemRandomization,disableRegisterRandomization"
)
println(getVerilogString(new Top, pretty))
```
Which generates:
```Verilog
 [info] welcome to sbt 1.7.1 (Oracle Corporation Java 17-ea)
[info] loading settings for project scastie3490744881740458038-build from plugins.sbt ...
[info] loading settings for project scastie3490744881740458038 from build.sbt ...

module FinalA(
  input   io_aaa, // @[src/main/scala/main.scala 8:14]
  output  io_bbb // @[src/main/scala/main.scala 8:14]
);
  assign io_bbb = io_aaa; // @[src/main/scala/main.scala 13:10]
endmodule
module Top(
  input   clock,
  input   reset,
  input   io_a, // @[src/main/scala/main.scala 21:14]
  output  io_b // @[src/main/scala/main.scala 21:14]
);
  wire  uMod_io_aaa; // @[src/main/scala/main.scala 26:20]
  wire  uMod_io_bbb; // @[src/main/scala/main.scala 26:20]
  FinalA uMod ( // @[src/main/scala/main.scala 26:20]
    .io_aaa(uMod_io_aaa),
    .io_bbb(uMod_io_bbb)
  );
  assign io_b = uMod_io_bbb; // @[src/main/scala/main.scala 28:8]
  assign uMod_io_aaa = io_a; // @[src/main/scala/main.scala 27:15]
endmodule
```
The input port of Module FinalA is still io_aaa, but it should be io_ccc.

**What is the current behavior?**
The input port of Module FinalA is still io_aaa, but it should be io_ccc.

**What is the expected behavior?**

io.aaa.suggestName("ccc") should rename `aaa` to `ccc`

**Please tell us about your environment:**

Scastie with Chisel 3.6.0-M1

**Other Information**

Have confirmed by @sequencer in the China Chisel Wechat group . 

**What is the use case for changing the behavior?**

mwachs5, 2023-01-05:
We don't support suggest name on bundle fields. For your use case can you make these two separate IO calls

mwachs5, 2023-01-05:
(Id like to make it an error, compile or runtime, to suggestName in things that will ignore it)

zhutmost, 2023-01-05:
Thanks for your response very much. 
Actually, I want the IO ports to be differently named for different cases. If `suggestName` is unavailable, which API should I use?

sequencer, 2023-01-05:
name in Bundle  is used to computed by reflection, and now is via plugin.
Which means it doesn't share the same naming strategy. You can directly use `Record` and override the `elements` to give name to fields in the `BundleType`.
And indeed, using `suggestName` on `Binding` that cannot be renamed should throw a `ChiselException`.