chipsalliance/chisel #2946: Should we allow bit-select on/with Bool() type?

mwachs5, 2023-01-22:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**

It is nonintuitive what it is supposed to mean if you have a Bool type and do a bit-select on it (which is apparently allowed even if the thing you are bit-selecting with is a boolean!). IN fact it is strange that we allow a Bool to be the selector too.

**Describe the solution you'd like**

It would make more sense if it were a compile or runtime error to do a bitselect on a Bool type or a bitSelect where the selector is a Bool type.

**Describe alternatives you've considered**

Leave things as they are

**Additional context**

https://scastie.scala-lang.org/HU5nisSDQR6g35rO66Nzhg

**What is the use case for implementing this feature?**

What is the use case for allowing this?

ingallsj, 2023-02-17:
> What is the use case for implementing this feature?

Catching bugs at compile time instead of silently allowing them.

mwachs5, 2023-02-17:
I think that making this a compile time error will be a heavy lift, but I think we can focus on a shorter-term runtime error/warning solution