chipsalliance/chisel #2961: the object PriorityEncoder may make something wrong in HDLbits site

fanliyong007, 2023-01-29:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
I use chisel3 to solve the [Problem](https://hdlbits.01xz.net/wiki/Always_case2) a  priority encoder.So I use PriorityEncoder.
The code as shown below.
<code>
class Alwayscase2  extends Module {
  val in=IO(Input(UInt(4.W)))
  val pos=IO(Output(UInt(2.W)))
  val pos_reg=Reg(UInt(2.W))
  pos:=pos_reg
  pos_reg:=PriorityEncoder(in)
}
</code>
It will generate as shown below.I have reduced something inneed
<code>
module top_module(
  input  [3:0] in,
  output  [1:0] pos
);
    always @(*) begin
    if (in[0]) begin // @[Mux.scala 47:70]
      pos <= 2'h0;
    end else if (in[1]) begin // @[Mux.scala 47:70]
      pos <= 2'h1;
    end else if (in[2]) begin // @[Mux.scala 47:70]
      pos <= 2'h2;
    end else begin
      pos <= 2'h3;
    end
  end
endmodule
</code>
but is wrong when the port 'in' is 4'b000 so I need to change the default  
<code>
module top_module(//this version can solve the problem
  input  [3:0] in,
  output reg [1:0] pos
);
    always @(*) begin
        if (in[0]) begin // @[Mux.scala 47:70]
      pos <= 2'h0;
    end else if (in[1]) begin // @[Mux.scala 47:70]
      pos <= 2'h1;
    end else if (in[2]) begin // @[Mux.scala 47:70]
      pos <= 2'h2;
    end else if (in[3]) begin // @[Mux.scala 47:70]
      pos <= 2'h3;
    end else begin
      pos <= 2'h0;
    end
  end
endmodule
</code>
**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

ekiwi, 2023-01-29:
The [documentation of the PriorityEncoder](https://www.chisel-lang.org/api/3.2.8/chisel3/util/PriorityEncoder$.html) states:
> Returns the bit position of the least-significant high bit of the input bitvector. 

So if your input is `0` then there is not high bit and thus the output is undefined. Maybe we should add a note to the documentation that says clearly that the input is expected to be one-hot encoded. What do you think?

fanliyong007, 2023-01-30:
> The [documentation of the PriorityEncoder](https://www.chisel-lang.org/api/3.2.8/chisel3/util/PriorityEncoder$.html) states:
>
>
> So if your input is `0` then there is not high bit and thus the output [...]

It's a  good idea.

ekiwi, 2023-01-30:
Let's keep this open until the documentation is improved.

closed 2023-02-03 by jackkoenig