chipsalliance/chisel #2985: Output bundle can't be assigned Mux

colin4124, 2023-02-08:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

```scala
import chisel3._
import chisel3.util._

class Top extends RawModule {
  val q = IO(Output(Vec(2, Bool())))

  val a = WireInit(VecInit(1.B, 0.B))
  val b = WireInit(VecInit(0.B, 1.B))

  q <> Mux(1.B, a, b)
}

object Main extends App {
  (new chisel3.stage.ChiselStage)
    .emitVerilog(new Top, args)
}
```

**What is the current behavior?**

```
chisel3.internal.ChiselException: internal error: attempted to generate LHS ref to ReadOnlyBinding OpBinding(Top@248deced,None)
```

**What is the expected behavior?**

It shouldn't error.

**Please tell us about your environment:**

- version: `3.6-SNAPSHOT`

**Other Information**

**What is the use case for changing the behavior?**

jackkoenig, 2023-02-09:
That's annoying

The good news is that this is fairly easy to work around, you can just use `:=` instead of `<>`. Muxes always require passive types (ie. not bidirectional) so you're not getting any additional value out of `<>` (but obviously it should not error).

colin4124, 2023-02-10:
Thanks,  `:=` works. I previously thought that  `Bundle` or Vec must use `<>` to be connected.

Maxpicca-Li, 2023-02-10:
I still have a question. When I use `chiseltest 0.3.2`, it is ok to compile when I use `<>`. But when I change the version of chiseltest from `0.3.2` to `0.5.2`, using `<>` will report the above error, that is: 
```shell
chisel3.internal.ChiselException: internal error: attempted to generate LHS ref to ReadOnlyBinding OpBinding(Top@248deced,None)
```
It will be troublesome to change all `<>` to `:=`. 
Can `<>` be compatibility in different versions of chiseltest?