chipsalliance/chisel #2998: Cannot use Record class to generate RegInit, errors are reported

yxysss, 2023-02-11:
I am trying to use Record class to generate RegInit. 
Here are my codes.

```
  class MyReg(reg_info: List[(String, UInt)]) extends Record {
    override def elements: ListMap[String, UInt] = ListMap(
      reg_info.map(
        x =>
          x._1 -> x._2
      ):_*
    )

    override def cloneType: MyReg.this.type = new MyReg(reg_info).asInstanceOf[this.type]
  }

class reg_test(reg_list: List[(String, String, List[(String, Int)])]) extends Module with RegSugar {
  val io = IO(new Bundle() {
    val a = Input(UInt(16.W))
    val b = Input(UInt(16.W))
    val d = Output(UInt(16.W))
    val e = Output(UInt(16.W))
  })

  val myBundle = new Bundle() {
    val a = UInt(16.W)
    val b = UInt(16.W)
  }
  val reg_init_bundle = RegInit(myBundle, 0.U.asTypeOf(myBundle))
  println(reg_init_bundle)
  reg_init_bundle.a := io.a
  reg_init_bundle.b := io.b

  val list_b = List(("a", UInt(16.W)), ("b", UInt(16.W)))
  val myRecord = new MyReg(list_b)

  val reg_init = RegInit(myRecord, 0.U.asTypeOf(myRecord))
  reg_init.elements("a") := io.a
  reg_init.elements("b") := io.b

  io.d := reg_init.elements("a") + reg_init.elements("b")
  io.e := reg_init_bundle.a + reg_init_bundle.b
}
```

In my sight, Bundle extends Record and if these code work on Bundle, they should also work on Record.
However, the error is reported on the line: 
```
val reg_init = RegInit(myRecord, 0.U.asTypeOf(myRecord))
```
```
Exception in thread "main" chisel3.package$RebindingException: Attempted reassignment of binding to reg_test.reg_init_?.a: Wire[UInt<16>], from: ChildBinding(reg_test.reg_init_?: Reg[MyReg])
	at ... ()
	at implicit_test.reg_test.$anonfun$reg_init$2(reg_test.scala:61)
	at chisel3.internal.prefix$.apply(prefix.scala:48)
	at implicit_test.reg_test.$anonfun$reg_init$1(reg_test.scala:61)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at implicit_test.reg_test.<init>(reg_test.scala:61)
	at implicit_test.reg_test$.$anonfun$verilogString$1(reg_test.scala:77)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
```

It seems that these codes can successfully generate RegInit with Bundle but failed to generate RegInit with Record.
Is that a way to generate RegInit with Record? or there is a compiler error?

jackkoenig, 2023-02-11:
This is a common sharp edge when using `Record` that is part of why it's considered more of an "advanced API". The basic problem is that Chisel needs the ability to create completely fresh clones of any `Data` (including `Record`) so that these clones can be "bound" as Regs, Wires, IOs, etc. This need for fresh clones is recursive, so when cloning a `Record` it needs to be able to properly clone any elements of that `Record`. In the implementation of `cloneType` in your code, you pass `reg_info` to the constructor. The problem is, the exact same UInt objects will be used for each clone of your `Record`. This violates Chisel's need for fresh objects. It manifests as an "Attempted reassignment of binding" because the exact same UInt that was an element of the literal in `0.U.asTypeOf(myRecord)` is used as an element of `reg_init`.

Historically for Records, Chisel has no way of checking this to give a better error message, because by API design, you defined the `cloneType` method--there's no hook for Chisel to do the check. Starting in Chisel 3.5.5 and enforced more heavily in Chisel 3.6 (which will be released soon), Chisel will generate cloneType methods for Records and will give a more precise error message. There is also a page on the next version of the website which will be live on chisel-lang.org after 3.6 is released, but there's sort of a pre-release here right now: http://chipsalliance.github.io/chisel3/chisel3/docs/cookbooks/cookbook.html#-how-do-i-deal-with-the-unable-to-clone-error 

Hopefully that explanation makes sense, the fix is to manually clone the arguments that are used for `val elements`, something like:
```scala
import chisel3.experimental.DataMirror
class MyReg(reg_info: List[(String, UInt)]) extends Record {
  override def elements: ListMap[String, UInt] = ListMap(
    reg_info.map(
      x =>
        // chiselTypeClone also clones set direction, whereas .cloneType does not
        x._1 -> DataMirror.internal.chiselTypeClone(x._2)
    ):_*
  )

  override def cloneType: MyReg.this.type = new MyReg(reg_info).asInstanceOf[this.type]
}
```

closed 2023-02-11 by jackkoenig

yxysss, 2023-02-12:
Great Thanks. That helps me a lot to understand the underlying logic.
If you write a Record and define cloneType by yourself, cloneType method should do deep copy on elements if there are Chisel Type fields.