chipsalliance/chisel #3057: Delete LegacyBlackBox

mwachs5, 2023-03-02:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**

We have an `abstract class LegacyBlackBox` which has comments on it that say "don't use this!" and I don't think we use it internally either. We can deprecate in 3.6 and delete in Chisel5

**Describe the solution you'd like**

delete this unused code

**Describe alternatives you've considered**

Keep it around? I think we don't have the legacy behavior anymore...?

**Additional context**

None

**What is the use case for implementing this feature?**

Not having unused APIs to support that say "don't use this!" in the comments.

17Moons, 2024-02-28:
i would like to work on this issue

jackkoenig, 2024-02-28:
Hello again @17Moons and thank you for your interest! It turns out this actually has already been done, so I will close this issue.

Fixed by https://github.com/chipsalliance/chisel/pull/3058.

closed 2024-02-28 by jackkoenig