chipsalliance/chisel #3084: A name followed by the same name with an trailing underscore results in `java.lang.NumberFormatException`

sterin, 2023-03-12:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

Construct hardware with some name, then construct additional hardware with the same name followed by an underscore. For example:

```scala
package foobar

import chisel3._

class Foo extends Module {
  val bar = RegInit(false.B)
  val bar_ = RegInit(false.B)
}

object EmitFoo {
  def main(args: Array[String]): Unit = {
    (new chisel3.stage.ChiselStage).emitVerilog(new Foo)
  }
}
```

**What is the current behavior?**

Fails, throws `java.lang.NumberFormatException`. 

```
[info] running foobar.EmitFoo 
[error] (run-main-0) java.lang.NumberFormatException: For input string: ""
[error] java.lang.NumberFormatException: For input string: ""
[error]         at ... ()
[error]         at ... ()
[error]         at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
```

**What is the expected behavior?**

Construct the appropriate hardware.

**Please tell us about your environment:**

Chisel version 3.5.6 and 3.6.0-RC2. Reproducible in official Scastie environment of 3.6.0-RC2.

https://scastie.scala-lang.org/jWaVv1vsSyGSG5I7Rvw9dw

**Other Information**

**What is the use case for changing the behavior?**

Chisel fails with a completely opaque error message.

azidar, 2023-03-13:
@seldridge I believe this is an MFC issue?

sterin, 2023-03-13:
@azidar It happens with Chisel 3.5.6 as well. I've added a 3.5.6 Scastie link to the issue. It is not clear from the stack trace, but it occurs inside the compiler plugin.