chipsalliance/chisel #3112: Memory Intrinsic API

sequencer, 2023-03-18:
**Memory** is not SRAM only, they should also include, SRAM, ROM, RF

**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
Chisel is using behavior memory since it was invented. However for all of the ASIC designs, the Memory IP is limited, even in FPGA, the choice of memory is also a big issue. designer is required to know what memory they are using before design, thus the current Mem/SyncMem API is quite anti-pattern for most of users: They define the memory, describe the memory behavior using `when` block, and let CIRCT infer which memory they are using. And most of user, including I have been struggling in frighting with it to generate the correct memory interface.

Thus I'm proposing adding a new Memory interface by directly define an memory `extmodule` with intrinsic, e.g. `class SRAM(nWord: Int, nBits: Int, replaceMacroname: String, ...)` it should directly generate `firrtl.mem` without any inference in CHIRRTL.

I do believe the currently memory behavior is harmful. Except senior engineer knowing how to define memory with the original API, I found most of newbies to chisel/RTL world are abusing it and think the memory can be abused like this.

Thus I propose after we finish the new Memory API, we should deprecate original API to make the chisel memory becoming reasonable.  
  
**Describe the solution you'd like**

Here are my proposal:
1. Internal review on memory IPs from different Fabs and FPGAs, give a list of popular ASIC/FPGA memories compiler and those behavior.
2. Conclude a general API for these memories, including define ports with the attribute of `R`, `W`, `RW`, and hazard map.(We can provide another library to predefine it)
4. Implement the chisel frontend and CIRCT backend.
5. Deprecate Mem and SyncMem

There are a lot of other benefits: we can simply define RF/ROM as one of Memory intrinsic, we may have better API to define the preset of memory(for FPGA) and ROM(for rom compiler), RF can be also defined with it, currently most of users are using a RF Module, and replace them in the PnR flow.

ekiwi, 2023-03-18:
I agree that the current way that `read` and `write` work is broken. I do wonder if we could fix it without completely removing `Mem` though. How about we add new methods like `createReadPort`, `createWritePort` and `createReadWritePort` that add a port to the memory that you can connect to like you would to a sub-module? We could enforce that these methods can only be called outside a `when` scope and thus they would be purely structural.

sequencer, 2023-03-19:
Think about it:
```scala
val a = Mem(???)
val p0 = a.createReadPort(???)
val p1 = a.createReadPort(???)
...
```
IMO, this API is so mutable that I wanna avoid, which just like this:
```scala
val a = Reg(???)
val aResetPort = a.addReset()
val aClockPort = a.addClock()
...
```

Does this feel like too Java?
since this is a breaking change, why not totally break it, and do the right thing directly? ;p

schoeberl, 2023-03-20:
+1 for explicit ports in memories. It could also be a configuration parameter at instantiation instead of createReadPort().

fzi-hielscher, 2023-03-22:
If you allow me to add my two cents, as I think this is a pretty important discussion and relates to several of my long held chisel pet peeves: I agree that the current API is impractical in many situations and I spent more time than I'd like to admit working around its limitations with custom wrapper classes, BlackBoxes and (most annoyingly) Python scripts. But on the other hand I would advise against going too low level and specifying more of the memory interface than what is strictly necessary for the design to work correctly. For example, I would not be happy finding a foundry specific macro hardcoded in a design. I feel that this unnecessarily breaks the separation between interface and implementation.

Most importantly, I would want to be able to easily switch between predefined and custom implementation generators without having to touch every occurrence of the respective memory. However, what aspects of this are supposed to be controlled by the CIRCT backend? Do we want different backends for different technologies or FPGA vendors? 

If anyone should be interested I can provide some more details on the issues and wishes I had regarding memories, targeting both FPGA and ASIC flows.