chipsalliance/chisel #3118: Support deduplication of modules where an output is being used as a source in BoringUtils.bore()

oharboe, 2023-03-19:
@jackkoenig @seldridge 

**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**

Sometimes it is useful to using BoringUtils.bore() with the source being an output of a deduplicated macro

**Describe the solution you'd like**

Deduplication should work.

**Describe alternatives you've considered**

Connecting to a wire in the module that instantiates the module that I want to bore to.

**Additional context**

https://scastie.scala-lang.org/KtzZQ3nFTea9KoNh0tRqtg

**What is the use case for implementing this feature?**

seldridge, 2023-03-20:
Here's a very reduced example:

```scala
//> using scala "2.13.10"
//> using repository "https://s01.oss.sonatype.org/content/repositories/snapshots"
//> using lib "org.chipsalliance::chisel::5.0.0-M1+32-baed66d4-SNAPSHOT"
//> using plugin "org.chipsalliance:::chisel-plugin::5.0.0-M1+32-baed66d4-SNAPSHOT"

import chisel3._
import chisel3.util.experimental.BoringUtils
import chisel3.experimental.{annotate, ChiselAnnotation}

import firrtl.transforms.MustDeduplicateAnnotation

object mustDedup {
  def apply(modules: RawModule*): Unit = {
    annotate(
      new ChiselAnnotation {
        override def toFirrtl =
          MustDeduplicateAnnotation(modules.map(_.toTarget))
      }
    )
  }
}

class Bar extends RawModule {
  val a = IO(Input(Bool()))
  val b = IO(Output(Bool()))
  b := a
}

class Foo extends RawModule {
  val a = IO(Input(Bool()))
  val b = IO(Output(Bool()))
  val c = IO(Output(Bool()))

  val bar1, bar2 = Module(new Bar)
  mustDedup(bar1, bar2)

  bar1.a := a
  bar2.a := bar1.b
  b := bar2.b
  BoringUtils.bore(bar2.b, Seq(c))
}

object Main extends App {
  println(
    circt.stage.ChiselStage.emitSystemVerilog(
      new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

This will then error out with:

```
# scala-cli BoringUtils.scala
Exception in thread "main" circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: firtool returned a non-zero exit code
------------------------------------------------------------------------------
ExitCode:
1
STDOUT:

STDERR:
<stdin>:2:1: error: module "Bar_1" not deduplicated with "Bar"
circuit Foo :
^
<stdin>:9:10: note: module marked NoDedup
  module Bar_1 :
         ^

------------------------------------------------------------------------------
```

The underlying issue comes from two things:

1. When a user does `BoringUtils.bore(...)` pointing at an instance it is entirely ambiguous what the user means. Do they mean the instance port or the module port?
2. The `SourceAnnotation`/`SinkAnnotation` representation of this is similarly ambiguous. This is currently encoded as a `ComponentName`, i.e., `<circuit>.<module>.<component>`. This would be able to differentiate module from instance if it was an actual `Target`. However, the problem of (1) still remains.

(1) is extremely fundamental and we should fix it in circuit. However, it is baked into the essence of Chisel. When you do `bar2.b`, this is pointing at the exact same object that is the module port. There is zero way to indicate that this is actually the instance port. (This is true even if context was captured. Just because the access was outside the module doesn't mean that the user intended to tap the instance port.)

Chisel needs to move away from the module and instance being the same object in order for this to actually work.

oharboe, 2023-03-20:
> Here's a very reduced example:
>
> ```scala
> //> using scala "2.13.10"
> //> using repository "https://s01.oss.sonatype.org/content/repositories/snapshots"
> //> using lib [...]

Thanks!