chipsalliance/chisel #3162: Add Explicit Memory APIs

seldridge, 2023-04-05:
Add generator APIs for constructing memories with an exact number of read, write, and readwrite ports. This enables designers to avoid odd behavioral rules and makes the Verilog generated match their intent.

For now, this can be a wrapper around a module (or just a class) that will use the existing memory inference rules so that it will produce the desired output. In the future this can be changed to directly building a FIRRTL memory or using a memory intrinsic.

ekiwi, 2023-04-05:
> For now, this can be a wrapper around a module (or just a class) that will use the existing memory inference rules so that it will produce the desired output. In the future this can be changed to directly building a FIRRTL memory or using a memory intrinsic.

I like this plan! As long as we get firrtl memories then all the existing simulator and formal verification infrastructure will continue working. For physical design, it would be cool if there was an easy way to allow users to replace memories from outside the RTL design code.

tianrui-wei, 2023-04-20:
There is an implementation at https://github.com/tianrui-wei/chisel-memory, where the default API is preserved while maintaining the ability to change the underlying implementation for different backend EDA tools.

jackkoenig, 2023-04-24:
@tianrui-wei can you add a license* to that repository?

*Ideally a permissive one

tianrui-wei, 2023-04-24:
Hi @jackkoenig, I've updated the repo with a bsd-3 license. Let me know if you want other implementations or some documentation. You can always reach me on matrix/email. Happy to PR things. :)