chipsalliance/chisel #3236: Adding a flag to `circt.stage.ChiselStage.emitFIRRTL` which presents the pre-optimized FIRRTL rather than the (post-) optimized FIRRTL

mmaloney-sf, 2023-05-03:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
I would like a way to inspect the FIRRTL output of my Chisel program's execution to verify it is working the way I am intending it to.

In particular, I would like to see the FIRRTL as it is presented to the CIRCT optimizer rather than the resulting FIRRTL that comes out at the end of the optimizer.

I know that the `emitCHIRRTL` method provides something very close to this -- and for the time being, that is a viable workaround. However, it seems that "CHIRRTL" is an informal extension of FIRRTL. It is arguably an implementation detail that has leaked into the public API.

I would like the guarantee that the output is valid FIRRTL, as per the spec, but before any optimization takes place.

**Describe the solution you'd like**

I think it would be simple to either have an optional `optimize` argument to the `emitFIRRTL` method, eg: `stage.emitFIRRTL(optimize=false)`.

Alternatively, it could be a separate method: `stage.emitUnoptimizedFIRRTL`.

**What is the use case for implementing this feature?**

I want this for pedagogical purposes.

When writing generators in Chisel, you have a lot of power in the kind of hardware you can generate. It is often instructive to inspect the FIRRTL output to get a more concrete idea of what the generator "does". I want to be able to talk about the FIRRTL output of a generator while avoiding:

1. A discussion of what "CHIRRTL" is, and
2. Having to compensate for the optimizer

seldridge, 2023-05-05:
If you want specification FIRRTL, this will need some work on the CIRCT side. CIRCT has an exporter of FIRRTL dialect, but it's a bit out of date. It can't export all of FIRRTL dialect anymore. If you are fine with FIRRTL dialect (MLIR) this is a bit easier and you don't have to use the exporter.

Roughly, this needs to run the following pipeline (which we could make easier to do in `firtool` natively):

```
firtool Qux.fir -parse-only | circt-opt -firrtl-lower-chirrtl | circt-translate -export-firrtl
```

zhutmost, 2023-05-14:
I think one doc page dedicated to `ChiselStage.emitXXX(***)` is very necessary. 
I don't find somewhere to detail the function/args of these APIs.

seldridge, 2023-05-14:
Fair point. The best documentation for this is the API docs: https://javadoc.io/doc/org.chipsalliance/chisel_2.13/latest/circt/stage/ChiselStage$.html

zhutmost, 2023-05-14:
> Fair point. The best documentation for this is the API docs: https://javadoc.io/doc/org.chipsalliance/chisel_2.13/latest/circt/stage/ChiselStage$.html

Thanks for your reply. The API doc is not enough. :(
From the API doc (i.e. the in-code comment doc), I cannot find what arguments I can pass to ChiselStage.emitSystemVerlogFile(gen, args, firtoolOpts). There is no doc which can tell me what “args” is allowed.

For example,  I can pass "-split-verilog" as one of `args` or `firtoolOpts`(They are both allowed, very strange), but the output files will be a bit different. That would be better if we can have a doc page for this ("How to control the RTL generation procedure").  :P