chipsalliance/chisel #3277: Provide a better message when firtool is not available in the path

carlosedp, 2023-05-12:
When running a module, Chisel should print a clearer message when `firtool` is not found.

Currently it is:

```
❯ ./mill chiselv.run
Compiling /home/carlosedp/repos/chiselv/build.sc
[38/51] chiselv.compile
Compiling compiler interface...
[info] compiling 18 Scala sources to /home/carlosedp/repos/chiselv/out/chiselv/compile.dest/classes ...
[info] done compiling
[51/51] chiselv.run
circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: firtool returned a non-zero exit code. Note that this version of Chisel (3.6.0) was published against firtool version 1.37.0.
------------------------------------------------------------------------------
ExitCode:
-1
STDOUT:

STDERR:

------------------------------------------------------------------------------

1 targets failed
chiselv.run subprocess failed
```

phatonix, 2023-05-22:
I have the same problem below:

Exception in thread "main" circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: firtool returned a non-zero exit code. Note that this version of Chisel (5.0.0) was published against firtool version 1.40.0.
------------------------------------------------------------------------------
ExitCode:
-1
STDOUT:

thebyrd, 2023-06-01:
I ran into the same issue. How did you get around it?

fangrouli, 2023-06-08:
> I ran into the same issue. I fixed it by downloading the firtool binary from the releases page of llvm/circt : https://github.com/llvm/circt/releases/tag/firtool-1.42.0
Hi! May I know which package you downloaded? Is it the .zip one?

thebyrd, 2023-06-13:
There's a binary in the zip/tarball. It needs to be in your path.

fangrouli, 2023-06-14:
> There's a binary in the zip/tarball. It needs to be in your path.

I see! thank you!

Pvview, 2023-10-27:
> I ran into the same issue. I fixed it by downloading the firtool binary from the releases page of llvm/circt : https://github.com/llvm/circt/releases/tag/firtool-1.42.0
> 
> Then making sure that binary was in my path

Thank for your answer !

jackkoenig, 2024-01-27:
This is fixed in Chisel 6.0.0 by use of [firtool-resolver](https://github.com/chipsalliance/firtool-resolver). Chisel now manages the binary of firtool for you.

closed 2024-01-27 by jackkoenig

DIGI1919, 2024-02-07:
> This is fixed in Chisel 6.0.0 by use of [firtool-resolver](https://github.com/chipsalliance/firtool-resolver). Chisel now manages the binary of firtool for you.

> There's a binary in the zip/tarball. It needs to be in your path.

hi! when i build the chisel project , i find that the firrtl is in the '.cache/coursier/v1/https/repo1.maven.org/maven2/edu/berkeley/cs/ ' ,and it's the .jar file, but the firtool is the bin file . how to do with firtool to make it join the project? thank you very much.

jackkoenig, 2024-02-07:
If you see `firrtl` in `edu/berkeley/cs`, that is the old Scala FIRRTL Compiler with an older version of Chisel (3.6 or earlier). Firtool is only handled for Chisel 6.0.0 and up.

DIGI1919, 2024-02-08:
> If you see `firrtl` in `edu/berkeley/cs`, that is the old Scala FIRRTL Compiler with an older version of Chisel (3.6 or earlier). Firtool is only handled for Chisel 6.0.0 and up. Can you provide more information about how you are using Chisel? eg. what version? Are you using an existing project like Chipyard or is this your own build using `SBT`, do you know what version of Chisel you are using?

   oh.i just follow the chisel sample of https://github.com/ucb-bar/riscv-mini.git. they use the chisel 5.0.0. and in their introduce,i only need just clone and make the project.but when they use the chisel 3.6, there is no error at all. and i meet the "version of Chisel (5.0.0) was published against firtool version 1.40.0." now.
   so, now i should simply download the firtool's bin of 1.40.0 ,and put it into '.cache/llvm-firtool/' and rebuild my project ,or i need some tool to build firtool in my project? if i want to use sbt to easily create new project next time ,how can i avoid the error like this?

the sbt now:
`ThisBuild / scalaVersion     := "2.13.7"
ThisBuild / version          := "2.5.0"
ThisBuild / organization     := "edu.berkeley.cs"

val chiselVersion = "5.0.0"
val chiseltestVersion = "5.0.2"

lazy val root = (project in file("."))
  .settings(
    name := "riscv-mini",
    libraryDependencies ++= Seq(
      "org.chipsalliance" %% "chisel" % chiselVersion,
      "edu.berkeley.cs" %% "chiseltest" % chiseltestVersion % "test"
    ),
    scalacOptions ++= Seq(
      "-language:reflectiveCalls",
      "-deprecation",
      "-feature",
      "-Xcheckinit",
    ),
    addCompilerPlugin("org.chipsalliance" % "chisel-plugin" % chiselVersion cross CrossVersion.full),
  )`

jackkoenig, 2024-02-08:
Oh I see, sorry for the confusion!

For Chisel 5.0.0, you should download [firtool version 1.40](https://github.com/llvm/circt/releases/tag/firtool-1.40.0) just put the firtool binary on your `PATH`. Unfortunately there's nothing you can do in SBT to solve this problem, you'll always need to make sure it's on your `PATH` when running Chisel (you could set this up in your `.bash_profile` or whatever for your shell).

This is much simpler in Chisel 6.0.0 since Chisel manages the version for you, I'm trying to see about backporting that improvement to 5.2.0 but it's a little bit tricky.

DIGI1919, 2024-02-09:
> Oh I see, sorry for the confusion!
>
> For Chisel 5.0.0, you should download [firtool version 1.40](https://github.com/llvm/circt/releases/tag/firtool-1.40.0) just put the firtool binary on your [...]

  thank you very much . but i still have no idea of the path where the firtool should be . i try put it in the '.cache/' and '.cache/llvm-firtool/' and rebuild the project ,but useless it is. so how can i realise the right path or how to set the path .

jackkoenig, 2024-02-09:
By `PATH`, I'm talking about the system environment variable. How you set it depends on what sort of system you are using and how you are invoking Chisel. Are you using Linux, MacOS, or Windows? If Linux, is it a shared machine? Are you using a shell (like bash) or an IDE (like VSCode)?

The typical workflow on Linux and MacOS would be to place the firtool binary in a reasonable location, eg. `/usr/local/bin`, so you could do something like:
```
curl -o - -L https://github.com/llvm/circt/releases/download/firtool-1.40.0/firrtl-bin-ubuntu-20.04.tar.gz | tar xf -
mv firtool-1.40.0/bin/firtool /usr/local/bin/
```

Alternatively, you could just add the directory containing firtool to your PATH,
```bash
export PATH=$PATH:$PWD/firtool-1.40.0/bin
```
But like I said, it depends on what kind of system you are on for the precise instructions.

DIGI1919, 2024-02-10:
> By `PATH`, I'm talking about the system environment variable. How you set it depends on what sort of system you are using and how you are invoking Chisel. Are you using Linux, MacOS, or Windows? If Linux, is it a shared machine? Are you using a shell (like bash) or an IDE (like VSCode)?
> 
> The typical workflow on Linux and MacOS would be to place the firtool binary in a reasonable location, eg. `/usr/local/bin`, so you could do something like:
> 
> ```
> curl -o - -L https://github.com/llvm/circt/releases/download/firtool-1.40.0/firrtl-bin-ubuntu-20.04.tar.gz | tar xf -
> mv firtool-1.40.0/bin/firtool /usr/local/bin/
> ```
> 
> Alternatively, you could just add the directory containing firtool to your PATH,
> 
> ```shell
> export PATH=$PATH:$PWD/firtool-1.40.0/bin
> ```
> 
> But like I said, it depends on what kind of system you are on for the precise instructions.

oh ! i get it .i put it in the /local/bin/ and it works. thank you very much.i think it work in the path of  sbt's things path which contains firrtl or chisels at first.