chipsalliance/chisel #3548: [Question] Why `-1.S.asUInt` in Mux compiled to `asUInt(asSInt(UInt<1>(1)))`

SpriteOvO, 2023-09-25:
I'm trying to compile this module

```scala
class LaneFFO(datapathWidth: Int) extends Module {
  // Seq(mask, data, destination)
  val src:          Vec[UInt] = IO(Input(Vec(3, UInt(datapathWidth.W))))
  val resultSelect: UInt = IO(Input(UInt(2.W)))
  val resp:         ValidIO[UInt] = IO(Output(Valid(UInt(datapathWidth.W))))
  val complete:     Bool = IO(Input(Bool()))
  val maskType:     Bool = IO(Input(Bool()))

  val truthMask: UInt = Mux(maskType, src.head, -1.S(datapathWidth.W).asUInt)
  val srcData:   UInt = truthMask & src(1)
  val notZero:   Bool = srcData.orR
  val lo:        UInt = scanLeftOr(srcData)
  // set before(right or)
  val ro: UInt = (~lo).asUInt
  // set including
  val inc: UInt = ro ## notZero
  // 1H
  val OH:    UInt = lo & inc
  val index: UInt = OHToUInt(OH)

  private def OH1ToOH(x:   UInt): UInt = (((x << 1): UInt) | 1.U) & ~Cat(0.U(1.W), x)
  private def OH1ToUInt(x: UInt): UInt = OHToUInt(OH1ToOH(x))
  resp.valid := notZero
  //
  val selectOH = UIntToOH(resultSelect)
  // find-first-set
  val first: Bool = selectOH(0)
  // set-before-first
  val sbf: Bool = selectOH(1)
  // set-only-first
  val sof: Bool = selectOH(2)
  // set-including-first
  val sif: Bool = selectOH(3)

  /** first:
    *   complete ? 0 : index
    * sbf:
    *   complete ? 0 : notZero ? ro : -1
    * sof:
    *   complete ? 0 : OH
    * sif:
    *   complete ? 0 : notZero ? inc : -1
    */
  val ffoResult: UInt = Mux1H(
    Seq(
      complete,
      !complete && first,
      !complete && notZero && sbf,
      !complete && sof,
      !complete && notZero && sif,
      !complete && !notZero && resultSelect(0)
    ),
    Seq(0.U, index, ro, OH, inc, -1.S.asUInt)
  )
  val resultMask: UInt = Mux(maskType && !first, src.head, -1.S(datapathWidth.W).asUInt)
  resp.bits := (ffoResult & resultMask) | (src.last & (~resultMask).asUInt)
}
```

It will be compiled to FIRRTL code

```firrtl
FIRRTL version 3.2.0
circuit LaneFFO :
  module LaneFFO :
    input clock : Clock
    input reset : UInt<1>
    input src : UInt<32>[3]
    input resultSelect : UInt<2>
    output resp : { valid : UInt<1>, bits : UInt<32>}
    input complete : UInt<1>
    input maskType : UInt<1>

    node _truthMask_T = asUInt(asSInt(UInt<32>(0hffffffff)))
    node truthMask = mux(maskType, src[0], _truthMask_T)
    node srcData = and(truthMask, src[1])
    node notZero = orr(srcData)
    node _lo_T = shl(srcData, 1)
    node _lo_T_1 = bits(_lo_T, 31, 0)
    node _lo_T_2 = or(srcData, _lo_T_1)
    node _lo_T_3 = shl(_lo_T_2, 2)
    node _lo_T_4 = bits(_lo_T_3, 31, 0)
    node _lo_T_5 = or(_lo_T_2, _lo_T_4)
    node _lo_T_6 = shl(_lo_T_5, 4)
    node _lo_T_7 = bits(_lo_T_6, 31, 0)
    node _lo_T_8 = or(_lo_T_5, _lo_T_7)
    node _lo_T_9 = shl(_lo_T_8, 8)
    node _lo_T_10 = bits(_lo_T_9, 31, 0)
    node _lo_T_11 = or(_lo_T_8, _lo_T_10)
    node _lo_T_12 = shl(_lo_T_11, 16)
    node _lo_T_13 = bits(_lo_T_12, 31, 0)
    node _lo_T_14 = or(_lo_T_11, _lo_T_13)
    node lo = bits(_lo_T_14, 31, 0)
    node ro = not(lo)
    node inc = cat(ro, notZero)
    node OH = and(lo, inc)
    node index_hi = bits(OH, 32, 32)
    node index_lo = bits(OH, 31, 0)
    node _index_T = orr(index_hi)
    node _index_T_1 = or(index_hi, index_lo)
    node index_hi_1 = bits(_index_T_1, 31, 16)
    node index_lo_1 = bits(_index_T_1, 15, 0)
    node _index_T_2 = orr(index_hi_1)
    node _index_T_3 = or(index_hi_1, index_lo_1)
    node index_hi_2 = bits(_index_T_3, 15, 8)
    node index_lo_2 = bits(_index_T_3, 7, 0)
    node _index_T_4 = orr(index_hi_2)
    node _index_T_5 = or(index_hi_2, index_lo_2)
    node index_hi_3 = bits(_index_T_5, 7, 4)
    node index_lo_3 = bits(_index_T_5, 3, 0)
    node _index_T_6 = orr(index_hi_3)
    node _index_T_7 = or(index_hi_3, index_lo_3)
    node index_hi_4 = bits(_index_T_7, 3, 2)
    node index_lo_4 = bits(_index_T_7, 1, 0)
    node _index_T_8 = orr(index_hi_4)
    node _index_T_9 = or(index_hi_4, index_lo_4)
    node _index_T_10 = bits(_index_T_9, 1, 1)
    node _index_T_11 = cat(_index_T_8, _index_T_10)
    node _index_T_12 = cat(_index_T_6, _index_T_11)
    node _index_T_13 = cat(_index_T_4, _index_T_12)
    node _index_T_14 = cat(_index_T_2, _index_T_13)
    node index = cat(_index_T, _index_T_14)
    connect resp.valid, notZero
    node selectOH = dshl(UInt<1>(0h1), resultSelect)
    node first = bits(selectOH, 0, 0)
    node sbf = bits(selectOH, 1, 1)
    node sof = bits(selectOH, 2, 2)
    node sif = bits(selectOH, 3, 3)
    node _ffoResult_T = eq(complete, UInt<1>(0h0))
    node _ffoResult_T_1 = and(_ffoResult_T, first)
    node _ffoResult_T_2 = eq(complete, UInt<1>(0h0))
    node _ffoResult_T_3 = and(_ffoResult_T_2, notZero)
    node _ffoResult_T_4 = and(_ffoResult_T_3, sbf)
    node _ffoResult_T_5 = eq(complete, UInt<1>(0h0))
    node _ffoResult_T_6 = and(_ffoResult_T_5, sof)
    node _ffoResult_T_7 = eq(complete, UInt<1>(0h0))
    node _ffoResult_T_8 = and(_ffoResult_T_7, notZero)
    node _ffoResult_T_9 = and(_ffoResult_T_8, sif)
    node _ffoResult_T_10 = eq(complete, UInt<1>(0h0))
    node _ffoResult_T_11 = eq(notZero, UInt<1>(0h0))
    node _ffoResult_T_12 = and(_ffoResult_T_10, _ffoResult_T_11)
    node _ffoResult_T_13 = bits(resultSelect, 0, 0)
    node _ffoResult_T_14 = and(_ffoResult_T_12, _ffoResult_T_13)
    node _ffoResult_T_15 = asUInt(asSInt(UInt<1>(0h1)))
    node _ffoResult_T_16 = mux(complete, UInt<1>(0h0), UInt<1>(0h0))
    node _ffoResult_T_17 = mux(_ffoResult_T_1, index, UInt<1>(0h0))
    node _ffoResult_T_18 = mux(_ffoResult_T_4, ro, UInt<1>(0h0))
    node _ffoResult_T_19 = mux(_ffoResult_T_6, OH, UInt<1>(0h0))
    node _ffoResult_T_20 = mux(_ffoResult_T_9, inc, UInt<1>(0h0))
    node _ffoResult_T_21 = mux(_ffoResult_T_14, _ffoResult_T_15, UInt<1>(0h0))
    node _ffoResult_T_22 = or(_ffoResult_T_16, _ffoResult_T_17)
    node _ffoResult_T_23 = or(_ffoResult_T_22, _ffoResult_T_18)
    node _ffoResult_T_24 = or(_ffoResult_T_23, _ffoResult_T_19)
    node _ffoResult_T_25 = or(_ffoResult_T_24, _ffoResult_T_20)
    node _ffoResult_T_26 = or(_ffoResult_T_25, _ffoResult_T_21)
    wire ffoResult : UInt<33>
    connect ffoResult, _ffoResult_T_26
    node _resultMask_T = eq(first, UInt<1>(0h0))
    node _resultMask_T_1 = and(maskType, _resultMask_T)
    node _resultMask_T_2 = asUInt(asSInt(UInt<32>(0hffffffff)))
    node resultMask = mux(_resultMask_T_1, src[0], _resultMask_T_2)
    node _resp_bits_T = and(ffoResult, resultMask)
    node _resp_bits_T_1 = not(resultMask)
    node _resp_bits_T_2 = and(src[2], _resp_bits_T_1)
    node _resp_bits_T_3 = or(_resp_bits_T, _resp_bits_T_2)
    connect resp.bits, _resp_bits_T_3
```

In the compiled FIRRTL code, I noticed the `-1.S.asUInt` for the `ffoResult` has been compiled to `asUInt(asSInt(UInt<1>(0h1)))`, but intuitively it should be `asUInt(SInt<32>(-1))`. It looks like it was given special treatment in CIRCT (or Chisel?) for some reason, but I've had no luck finding it and figuring it out. I would appreciate it if someone could explain this for me.

sequencer, 2023-09-25:
`circt.stage.ChiselStage.emitCHIRRTL` use firtool to parse the `.fir` generated by chisel. So this question might relate to circt parser.

sequencer, 2023-09-25:
My guess: https://github.com/llvm/circt/blob/249ae37e8821a6c459764a25618144c5defdb0a5/lib/Dialect/FIRRTL/Import/FIRParser.cpp#L2278

seldridge, 2023-09-25:
I think this is just how `-1.S.asUInt` will be emitted from Chisel. You can get the same thing with a simpler example:

```scala
//> using scala "2.13.11"
//> using lib "org.chipsalliance::chisel::6.0.0-M3"
//> using plugin "org.chipsalliance:::chisel-plugin::6.0.0-M3"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import circt.stage.ChiselStage

class Foo extends Module {
  val a = IO(Output(UInt()))

  a := -1.S.asUInt
}

object Main extends App {
  println(
    ChiselStage.emitCHIRRTL(new Foo)
  )
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

This produces the following FIRRTL text:
```
FIRRTL version 3.1.0
circuit Foo :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    output a : UInt

    node _a_T = asUInt(asSInt(UInt<1>(0h1)))/3548.scala 12:13]
    connect a, _a_T
```

CIRCT doesn't touch this in the parser, but will rip this out in the canonicalizer see:

Only parsing with `firtool -parse-only 3548.fir | circt-translate -export-firrtl`:
```
FIRRTL version 3.2.0
circuit Foo :
  module Foo : @[<stdin> 3:5]
    input clock : Clock @[<stdin> 3:27]
    input reset : UInt<1> @[<stdin> 3:53]
    output a : UInt @[<stdin> 3:82]

    node _a_T = asUInt(asSInt(UInt<1>(1))) @[<stdin> 7:15]
    connect a, _a_T @[<stdin> 8:7]
```

Parsing an canonicalizing with `firtool -parse-only 3548.fir | circt-opt -canonicalize | circt-translate -export-firrtl`:
```
FIRRTL version 3.2.0
circuit Foo :
  module Foo : @[<stdin> 3:5]
    input clock : Clock @[<stdin> 3:27]
    input reset : UInt<1> @[<stdin> 3:53]
    output a : UInt @[<stdin> 3:82]

    node _a_T = UInt<1>(1) @[<stdin> 5:15]
    connect a, _a_T @[<stdin> 6:7]
```

sequencer, 2023-09-25:
why:
```
FIRRTL version 3.1.0
circuit Foo :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    output a : UInt

    node _a_T = asUInt(asSInt(UInt<1>(0h1)))
    connect a, _a_T
```
has the width of 1?

sequencer, 2023-09-25:
Oh my...
https://github.com/chipsalliance/chisel/blob/7cc6d51a6c5ca21b96076f4af8eb21f64d0ae809/core/src/main/scala/chisel3/internal/firrtl/Converter.scala#L88-L92
Why???

SpriteOvO, 2023-09-25:
Oh, so I figured out the underlying reason.

https://github.com/chipsalliance/chisel/blob/7cc6d51a6c5ca21b96076f4af8eb21f64d0ae809/firrtl/src/main/scala/firrtl/ir/IR.scala#L215

Chisel uses `scala.math.BigInt(constant).bitLength` to set the width for int constant, and I didn't realize the standard function, so I wrote my own `bitLength` function:

```scala
def bitLength(n: Int): Int = {
  var bits = 0
  var num = n
  while (num != 0) {
    bits += 1
    num >>>= 1
  }
  max(bits, 1)
}
```

For `-1`, Scala `BigInt(constant).bitLength` returns `0` as the result, then expand to `1` by `max(bitLength, 1)`, and my own `bitLength` function returns `32` as the result, then caused the problem.

@seldridge Thanks for your hints.

closed 2023-09-25 by SpriteOvO