chipsalliance/chisel #3557: Boring from parent module with probe doesn't get a necessary :#= assignment

mwachs5, 2023-09-27:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

See the test in #3556. Even if fixing the issue reported there, the test will not pass, because of this picture:
[image]

**What is the current behavior?**
Note the bottom arrow in the picture: when we are goign straight from a bidirectional thing, and then turning it into an Input, we dont have any part of the code that does the stripping of direction and then using a `:#=` operator to drive the stripped-direction input to the bidirectional "real thing".

When there is a layer of hierarchy in the middle:
[image]

we have to first create the probe, which handles the stripping and the define works without the `#:=`.

**What is the expected behavior?**

We need to not emit a `connect` from a bidirectional thing to a monodirectional thing.

This can be done by either emitting the individual fields (as in the `:#=` would handle) or by making a probe define and then read in the same layer of hierarchy.

The former is a little tricky to implement because we can't directly use the `:#=` operator in the boring APIs, we can only add commands to the secretCommands, and there is no direct translation from `:#=` to the commands we would want to add to secretCommands.

It might be simpler to just always make the probe, define, and read even if we are in the same layer of heirarchy.

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

Need to get code compiling. There is a possible workaround -- users can make the Wire(Output(...)) themself in the source of the tap.

dtzSiFive, 2023-10-04:
> It might be simpler to just always make the probe, define, and read even if we are in the same layer of heirarchy.

This should be optimized away by CIRCT, if it's not LMK :+1: .  It does seem like it may simplify things a bit.

In a sense a `read(probe(X))` would become the implementation of a hypothetical `asPassive(X)` command, is that about right?

All good, but curious/interested: what's the reason/challenges preventing having `:#=` as a (sequence of) "secret command"?

seldridge, 2023-10-04:
> All good, but curious/interested: what's the reason/challenges preventing having :#= as a (sequence of) "secret command"?

It's due to current implementation. This can be changed. The problem is that `:#=` eventually results in calls to `:=` [here](https://github.com/chipsalliance/chisel/blob/6dbbc73c793bdd1a357a4088abfe4704f23195a8/core/src/main/scala/chisel3/connectable/Connection.scala#L124C30-L124C30). This will push commands to the module's normal command area. Instead, we have a problem where we need to push arbitrary commands. However, the module is currently only allowing you to create a `secretConnection` through a [restricted API](https://github.com/chipsalliance/chisel/blob/6dbbc73c793bdd1a357a4088abfe4704f23195a8/core/src/main/scala/chisel3/RawModule.scala#L186) that only works with a LHS and RHS Data (which then pushes commands into the secret command array).

To be able to use `:#=` there needs to be a way to tell any connection what buffer to add commands to. If we're then starting to add arguments to `:#=` and friends related to what buffer to push to, I don't see the point in maintaining two separate buffers.

tl;dr: there are some infrastructure changes that have to happen.

commit 2afe459e (2024-05-23): Fix boring tap of non-passive source from parent.

Expected result is fully aligned, which is what happens
when reading from a probe.

When the source is already at the LCA (from parent)
there's no probe and the secret connections only support
a few commands.

For lack of a way to do, e.g., `:#=` here,
use `read(probe(x))` to get the fully aligned result
that's expected and bored down to the child.

This isn't ideal but fixes this using only the sorts of expressions
and commands that we've already committed to supporting.

Fixes #3557.
PR #4083 (2024-05-24): Fix boring tap of non-passive source from parent.
Expected result is fully aligned, which is what happens when reading from a probe.

When the source is already at the LCA (from parent) there's no probe and the secret connections only support a few commands.

For lack of a way to do, e.g., `:#=` here,
use `read(probe(x))` to get the fully aligned result that's expected and bored down to the child.

This isn't ideal but fixes this using only the sorts of expressions and commands that we've already committed to supporting.

Fixes #3557.

### Contributor Checklist

- [x] Did you add Scaladoc to e [... cut]
commit 0dbedc31 (2024-05-24): Fix boring tap of non-passive source from parent. (#4083)

Expected result is fully aligned, which is what happens
when reading from a probe.

When the source is already at the LCA (from parent)
there's no probe and the secret connections only support
a few commands.

For lack of a way to do, e.g., `:#=` here,
use `read(probe(x))` to get the fully aligned result
that's expected and bored down to the child.

This isn't ideal but fixes this using only the sorts of expressions
and commands that we've already committed to supporting.

Fixes #3557.
commit afd7bfc7 (2024-05-24): Fix boring tap of non-passive source from parent. (#4083)

Expected result is fully aligned, which is what happens
when reading from a probe.

When the source is already at the LCA (from parent)
there's no probe and the secret connections only support
a few commands.

For lack of a way to do, e.g., `:#=` here,
use `read(probe(x))` to get the fully aligned result
that's expected and bored down to the child.

This isn't ideal but fixes this using only the sorts of expressions
and commands that we've already committed to supporting.

Fixes #3557.

(cherry picked from commit 0dbedc3154b6163061b65d [... cut]
closed 2024-05-24 by jackkoenig