chipsalliance/chisel #3567: Improve behavior with (RW)ProbeValue of literal

dtzSiFive, 2023-10-04:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

Write code including constructs such as `RWProbeValue(false.B)` and `ProbeValue(false.B)`.

**What is the current behavior?**

No error, produces invalid probe expressions, which must target a reference.

Presently:

```firrtl
rwprobe(UInt<1>(0h0))
```

and

```
probe(UInt<1>(0h0))
```

are produced, both of which are invalid.

These will be rejected by firtool.

(This may have come up previously but couldn't find issue about it)

**What is the expected behavior?**

Either materialize a node to hold the probed value (more for `probe`), or error.
RWProbe of a literal seems reasonable to reject.

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

Better experience using probes.

debs-sifive, 2023-10-31:
@dtzSiFive can you take a look at https://github.com/chipsalliance/chisel/pull/3610 and confirm if it addresses this issue fully?

dtzSiFive, 2023-11-28:
I believe this has been addressed, thank you!

closed 2023-11-28 by dtzSiFive