chipsalliance/chisel #3591: Make Chisel5 generate parameter in Verilog

yxysss, 2023-10-20:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
I think that it will be greater if chisel5 can add parameter function.
In Verilog, it can define parameter in Module.
In Chisel, can we still generate parameters in Module?

**Describe the solution you'd like**
Since Chisel5 has changed its firrtl compiler from SFC to MFC, I think adding parameter function would make Chisel5 even more powerful.

**What is the use case for implementing this feature?**
Consider this condition in which we can only add a parameter file in Verilog with statement `include.
As a result, we do not need to re-run Chisel5 to generate Verilog if we only change the value of parameter in Chisel5.
I think this will be a great benefit to save the time when we use Chisel5 to generate Verilog.

Please consider this suggestion carefully.
Best Regards

sequencer, 2023-11-05:
I don't think this will happen in the near future.
However Chisel supports embed Parameter embedded as `Property` inside MLIR.

yxysss, 2023-11-22:
Thanks for your hint.
Is there a example about MLIR's Property?
Great thanks.

seldridge, 2023-11-22:
There are some hints of this in Chisel 6 and beyond. Something like this will happen. However, it is not expected to be where Scala generator-time parameters are automatically translated into Verilog parameters. Instead, this will be intentional in Chisel and not automatic.

I.e., expect to see parameters that enable optional or alternative behavior. Parameters associated with widths will also likely show up.

### Examples

#### Layers

Chisel 6 adds "layers" (formerly groups). This is a mechanism to declare optional verification functionality that can be enabled by including a file in your simulator compile. This is essentially a "parameter", but it is not a "Verilog parameter" even though we could envision a lowering using actual Verilog parameters in the future.

See: https://www.chisel-lang.org/docs/explanations/layers

#### Variants or Multiple Instances

Work is ongoing in CIRCT to add infrastructural support for and soon in the FIRRTL spec to add the ability to instantiate one of several modules. We're not sure exactly what this is going to look like in Chisel, yet. However, this is yet-another kind of parameter.

#### Parametric Widths and Instance Parameters

FIRRTL models external modules (Chisel's `BlackBox` or `ExtModule`) as binding the Verilog parameter values into the external module definition. I've long wanted to move this to the instantiation site. However, this requires understanding how the parameters affect the external module port widths (if they do). Sorting this out will help pipe clean how true parameters can be passed around to external modules, but not yet used.

See: https://github.com/chipsalliance/firrtl-spec/pull/46