chipsalliance/chisel #3606: Error import chisel3._

Guest-615695028, 2023-10-26:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
In Ubuntu Terminal,
```
szu@LAPTOP-T18N4BB4:~/chisel-empty/src/main/scala/empty$ sbt test
[warn] No sbt.version set in project/build.properties, base directory: /home/szu/chisel-empty/src/main/scala/empty
[info] welcome to sbt 1.9.7 (Private Build Java 19.0.2)
[info] set current project to empty (in build file:/home/szu/chisel-empty/src/main/scala/empty/)
[info] compiling 1 Scala source to /home/szu/chisel-empty/src/main/scala/empty/target/scala-2.12/classes ...
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:10:8: not found: object chisel3
[error] import chisel3._
[error]        ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:11:8: not found: object chisel3
[error] import chisel3.util._
[error]        ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:13:19: illegal inheritance from final class Module
[error] class Add extends Module {
[error]                   ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:14:12: not found: value IO
[error]   val io = IO(new Bundle {
[error]            ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:14:19: not found: type Bundle
[error]   val io = IO(new Bundle {
[error]                   ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:15:13: not found: value Input
[error]     val a = Input(UInt(8.W))
[error]             ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:15:19: not found: value UInt
[error]     val a = Input(UInt(8.W))
[error]                   ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:15:26: value W is not a member of Int
[error]     val a = Input(UInt(8.W))
[error]                          ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:16:13: not found: value Input
[error]     val b = Input(UInt(8.W))
[error]             ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:16:19: not found: value UInt
[error]     val b = Input(UInt(8.W))
[error]                   ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:16:26: value W is not a member of Int
[error]     val b = Input(UInt(8.W))
[error]                          ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:17:13: not found: value Output
[error]     val c = Output(UInt(8.W))
[error]             ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:17:20: not found: value UInt
[error]     val c = Output(UInt(8.W))
[error]                    ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:17:27: value W is not a member of Int
[error]     val c = Output(UInt(8.W))
[error]                           ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:20:13: not found: value RegInit
[error]   val reg = RegInit(0.U(8.W))
[error]             ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:20:23: value U is not a member of Int
[error]   val reg = RegInit(0.U(8.W))
[error]                       ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:20:27: value W is not a member of Int
[error]   val reg = RegInit(0.U(8.W))
[error]                           ^
[error] /home/szu/chisel-empty/src/main/scala/empty/Add.scala:28:3: not found: value emitVerilog
[error]   emitVerilog(new Add(), Array("--target-dir", "generated"))
[error]   ^
[error] 18 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 2 s, completed Oct 26, 2023, 11:24:37 AM
```
**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

edwardcwang, 2023-10-26:
Did you use https://github.com/freechipsproject/chisel-template? It looks like the Chisel libraries aren't present.

Martoni, 2023-11-12:
@edwardcwang the template  https://github.com/freechipsproject/chisel-template use chisel3 no ?
According to github Release the latest is 5.0.0

mwachs5, 2023-11-12:
The chisel template is using 5.0.0: https://github.com/freechipsproject/chisel-template/blob/main/build.sbt#L7

The package import chisel3 hasn't changed (yet).

closed 2024-01-27 by jackkoenig