chipsalliance/chisel #3608: Firtool Version compatibility issues

Pvview, 2023-10-29:
https://github.com/chipsalliance/chisel/issues/3277

**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Firtool Version compatibility issues. I'm always frustrated when i run riscv-mini.

**Describe the solution you'd like**
I'm on Ubuntu 22.04 . when i run riscv-mini with chisel 5 ,a compatibility issue occurs.
*************************************************
ExitCode:
-1
STDOUT:
*************************************************
STDERR:

**What is the use case for implementing this feature?**
I fixed it by downloading the firtool binary from the releases page of llvm/circt : https://github.com/llvm/circt/releases/tag/firtool-1.42.0. https://github.com/llvm/circt/releases/tag/firtool-1.42.0 Then making sure that binary was in my path

f-zl, 2023-11-04:
I have a related request. firtool doesn't release Windows executable before 1.45.0. But firtool 1.45.0 seems to generate system verilog by default (generate .sv when exeuting chisel3.emitVerilog). Can chisel support newer version of firtool?

sequencer, 2023-11-04:
Chisel main branch is always supports the up to date version of firtool.

jackkoenig, 2024-01-27:
As of Chisel 6.0.0 (and I hope to eventually backport similar support to 5.x and maybe even 3.6.x), Chisel now manages the version of firtool for you.

closed 2024-01-27 by jackkoenig