chipsalliance/chisel #3666: Differences Between := and :<*= involving Analog and DontCare

seldridge, 2023-12-01:
I've noticed that driving an `Analog` with a `DontCare` in different ways results in differences in whether the analog can be later attached. 

Basically, this is an error:

``` scala
b :<= DontCare
b :<= a
```

Whereas this is not an error:

``` scala
b := DontCare
b :<= a
```

The latter produces the following FIRRTL (which should probably elide the invalidate), but compiles:

```
    invalidate b
    attach (b, a)
```

Full example for playing around with:

``` scala
//> using scala "2.13.11"
//> using repository sonatype-s01:snapshots
//> using lib "org.chipsalliance::chisel::6.0.0-M3+117-2372b1c4-SNAPSHOT"
//> using plugin "org.chipsalliance:::chisel-plugin::6.0.0-M3+117-2372b1c4-SNAPSHOT"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import chisel3.experimental.Analog
import circt.stage.ChiselStage

class Foo extends Module {
  val a = IO(Analog(1.W))
  val b = IO(Analog(1.W))
  val c = IO(Analog(1.W))

  b := DontCare
  b :<= a

  c :<= DontCare
  c :<= a
}

object Main extends App {
  println(
    ChiselStage.emitCHIRRTL(
      gen = new Foo
    )
  )
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

mwachs5, 2023-12-01:
What is the error?

seldridge, 2023-12-01:
@mwachs5 wrote:
> What is the error?

Edited it into the issue text at the bottom.

sequencer, 2023-12-03:
Basically I think this is a pitfall in chisel type system, in chisel it runtime modify the chisel type when a wire is connect to other wire. 

Basically I think we need to think about embed all type information, here is the the Connectable semantic in FIRRTL spec and emit the connect IR, let firtool handle those errors. We should consider stopping detect these errors, make them a part of firrtl-spec and let a **real** compiler handle it.