chipsalliance/chisel #3677: Chisel5.1.0 cannot emit verilog

isjcliu, 2023-12-06:
chisel5.1.0 firtool 1.4.3
In the source code, the emitverilog function is a systemverilog instead of verilog.  I changed it to verilog, but the generated .v file is not available in vivado and contains the syntax of systemverilog

Gary2342, 2024-02-19:
I have the same problem. Do you find the way to solve this? 
I find the answer ( #3706 )about this problems saying chisel can only generate systemverilog?

Emin017, 2024-02-29:
You can add firtool [option](https://circt.llvm.org/docs/VerilogGeneration/) to convert unwanted system verilog syntax

Gary2342, 2024-02-29:
已收到

zhongwujie, 2024-04-22:
To avoid the generation of systemverilog syntax like "automatic logic", you can add
```
firtoolOpts = Array("-lowering-options=disallowLocalVariables")
```

Gary2342, 2024-04-22:
已收到