chipsalliance/chisel #3689: firtool 1.61.0 failes on "Unknown command line argument '-dedup'" when generating verilog

pineapplehunter, 2023-12-15:
**Type of issue**: Bug Report

When I try to generate a verilog source, firtool 1.61.0 exits with an error.
I see that this seems to be fixed on #3634.
Can a new version of chisel be released with this patch?

**Please provide the steps to reproduce the problem:**

Generate verilog with firtool 1.61.0

**What is the current behavior?**

```shell
...
(some build outputs)
...
[error] circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: firtool returned a non-zero exit code. Note that this version of Chisel (5.1.0) was published against firtool version 1.43.0.
[error] ------------------------------------------------------------------------------
[error] ExitCode:
[error] 1
[error] STDOUT:
[error] 
[error] STDERR:
[error] firtool: Unknown command line argument '-dedup'.  Try: 'firtool --help'
[error] firtool: Did you mean '--no-dedup'?
[error] 
[error] ------------------------------------------------------------------------------
[error] stack trace is suppressed; run last Compile / run for the full output
[error] (Compile / run) circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: firtool returned a non-zero exit code. Note that this version of Chisel (5.1.0) was published against firtool version 1.43.0.
[error] ------------------------------------------------------------------------------
[error] ExitCode:
[error] 1
[error] STDOUT:
[error] 
[error] STDERR:
[error] firtool: Unknown command line argument '-dedup'.  Try: 'firtool --help'
[error] firtool: Did you mean '--no-dedup'?
[error] 
[error] ------------------------------------------------------------------------------
[error] Total time: 8 s, completed 2023/12/16 2:46:30
```

**What is the expected behavior?**

It should generate a verilog file.

**Please tell us about your environment:**

- version: 5.1.0
- firtool: firtool-1.61.0
- OS: NixOS 24.05

**Other Information**

**What is the use case for changing the behavior?**

jackkoenig, 2023-12-15:
Chisel 6.0.0-RC1 works with firtool-1.61.0, please try it out!

pineapplehunter, 2023-12-16:
Thanks!
Generating verilog seems to be working.
But now I am having some issues with chiseltest.

```shell
...
[info]   java.lang.NoClassDefFoundError: firrtl/ir/PartialConnect
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at scala.collection.immutable.Vector2.map(Vector.scala:2140)
[info]   at scala.collection.immutable.Vector2.map(Vector.scala:443)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$1(ChiselBridge.scala:155)
[info]   at scala.collection.immutable.List.map(List.scala:246)
[info]   at scala.collection.immutable.List.map(List.scala:79)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:155)
[info]   ...
[info]   Cause: java.lang.ClassNotFoundException: firrtl.ir.PartialConnect
[info]   at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:445)
[info]   at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:103)
[info]   at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:588)
[info]   at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at scala.collection.immutable.Vector2.map(Vector.scala:2140)
[info]   at scala.collection.immutable.Vector2.map(Vector.scala:443)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   ...
```

seldridge, 2024-01-10:
This went through a deprecation schedule.

- This was deprecated in `firtool-1.57.0`: https://github.com/llvm/circt/pull/6188
- This option was removed in `firtool-1.60.0`: https://github.com/llvm/circt/pull/6191

The error message that you get for this tells you what version of `firtool` Chisel was built with, i.e., 1.43.0. The point of this message is to indicate that you should try using that `firtool` version.

Chisel 6 will manage your `firtool` installation for you and avoids that problem entirely.

gonsolo, 2024-01-10:
> The error message that you get for this tells you what version of firtool Chisel was built with, i.e., 1.43.0. The point of this message is to indicate that you should try using that firtool version.

That was what I did in the end. My point was that the deprecation period should be until the next Chisel release.

> Chisel 6 will manage your firtool installation for you and avoids that problem entirely.

That's of course the best solution. :)

DarkKirb, 2024-02-17:
I am trying out 6.0 since that is built for 1.61.0 but the automatic management of firtool installations is breaking chisel for me:

```
[error] (Compile / run) circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: /home/darkkirb/.cache/llvm-firtool/1.62.0/bin/firtool returned a non-zero exit code. Note that this version of Chisel (6.0.0) was published against firtool version 1.62.0.
[error] ------------------------------------------------------------------------------
[error] ExitCode:
[error] 127
[error] STDOUT:
[error] 
[error] STDERR:
[error] Could not start dynamically linked executable: /home/darkkirb/.cache/llvm-firtool/1.62.0/bin/firtool
[error] NixOS cannot run dynamically linked executables intended for generic
[error] linux environments out of the box. For more information, see:
[error] https://nix.dev/permalink/stub-ld
```

there is a firrtl of the correct version in $PATH, is there a way to use that over the nonfunctional “automatic” version?

jackkoenig, 2024-02-17:
> there is a firrtl of the correct version in $PATH

As mentioned under [API Modification in the v6.0.0 release notes](https://github.com/chipsalliance/chisel/releases/tag/v6.0.0), you should set `CHISEL_FIRTOOL_PATH` to point to the directory containing your firtool binary to override the automatically managed version.

> > NixOS cannot run dynamically linked executables intended for generic
>
> is there a way to use that over the nonfunctional “automatic” version?

I would argue that Nix is the nonfunctional bit here for violating POSIX 🙂

sequencer, 2024-02-18:
For Nix users:
Here our flow: build jar and wrap it. Set env to use circt from nixpkgs. 
```
  installPhase = ''
    runHook preInstall
    strip-nondeterminism elaborator.jar
    mkdir -p $out/share/java
    mv elaborator.jar $out/share/java/elaborator.jar
    mkdir $out/bin
    makeWrapper ${jre}/bin/java $out/bin/elaborator \
      --add-flags "-jar $out/share/java/elaborator.jar" \
      --set PATH ${sv-lang}/bin:${espresso}/bin \
      --set CHISEL_FIRTOOL_PATH ${circt}/bin
    runHook postInstall
  '';
```

sequencer, 2024-02-18:
BTW, `nixpkgs#circt` is also maintained by my team. Feel free to submit any bug reports. ;p

DarkKirb, 2024-02-18:
> I would argue that Nix is the nonfunctional bit here for violating POSIX 🙂

POSIX makes no no claims on whether or not you can run binaries compiled for a different POSIX operating system. As an example, you can’t run macOS binaries on AIX (both posix certified).

Additionally GNU libraries and tools are posix-incompatible in several spots so you can’t even defer to POSIX here

DarkKirb, 2024-02-18:
but thank you to pointing to the environment variable here at least

pineapplehunter, 2024-02-24:
The suggestions seem to be working! Thanks!
I saw the instructions to use chiseltest 6.0-SNAPSHOT written here https://github.com/ucb-bar/chiseltest/issues/699

With my NixOS setup, I got it to work with a dev shell like this.

```nix
mkShell {
  packages = [ ... ];
  CHISEL_FIRTOOL_PATH = "${circt}/bin";
}
```

This method works with mkDerivation too.

closed 2024-02-24 by pineapplehunter