chipsalliance/chisel #3735: Inconsistent behavior about zero-width operands

SpriteOvO, 2024-01-13:
**Type of issue**: Bug Report

```scala
class ZeroWidthTest extends Module {
  val a = IO(Input(UInt(32.W)))
  val b = IO(Input(SInt(32.W)))
  val c = IO(Output(UInt(32.W)))
  c := Cat(a >> 32, b)
}
```

**What is the current behavior?**

`c := Cat(a >> 32, b)` is represented as `UInt(32.W) := Cat(UInt(0.W), UInt(32.W))` in Chisel IR

**What is the expected behavior?**

In CIRCT MLIR, it represented as `UInt(33.W) := Cat(UInt(1.W), UInt(32.W))`, the first operand of the `Cat` is 1 width instead of zero-width.

```mlir
module {
  firrtl.circuit "ZeroWidthTest" {
    firrtl.module @ZeroWidthTest(in %clock: !firrtl.clock, in %reset: !firrtl.uint<1>, in %a: !firrtl.uint<32>, in %b: !firrtl.sint<32>, out %c: !firrtl.uint<32>) attributes {convention = #firrtl<convention scalarized>} {
      %0 = firrtl.shr %a, 32 : (!firrtl.uint<32>) -> !firrtl.uint<1>
      %_c_T = firrtl.node interesting_name %0 : !firrtl.uint<1>
      %1 = firrtl.asUInt %b : (!firrtl.sint<32>) -> !firrtl.uint<32>
      %c_lo = firrtl.node interesting_name %1 : !firrtl.uint<32>
      %2 = firrtl.cat %_c_T, %c_lo : (!firrtl.uint<1>, !firrtl.uint<32>) -> !firrtl.uint<33>
      %_c_T_1 = firrtl.node interesting_name %2 : !firrtl.uint<33>
      %3 = firrtl.tail %_c_T_1, 1 : (!firrtl.uint<33>) -> !firrtl.uint<32>
      firrtl.strictconnect %c, %3 : !firrtl.uint<32>
    }
  }
}
```

**What is the use case for changing the behavior?**

It caused problems in Binder, when we visit the `connect` in the example we get IR `connect UInt(32.W), UInt(32.W)`, therefore there is no way to add a `tail` op like CIRCT does, as the width of the 2 operands are the same.

sequencer, 2024-01-13:
Basically, this problem was coming from the firtool parser did too much things, binder should not depend on the width inference result(but firtool can).

SpriteOvO, 2024-01-20:
According to the FIRRTL spec:

```
## Shift Right Operation

  --------------------------------------------------------------------------
  Name   Arguments   Parameters   Arg Types   Result Type   Result Width
  ------ ----------- ------------ ----------- ------------- ----------------
  shr    \(e\)       \(n\)        (UInt)      UInt          max(w~e~-n, 1)

                                  (SInt)      SInt          max(w~e~-n, 1)
  --------------------------------------------------------------------------

The shift right operation truncates the least significant n bits from e.
If n is greater than or equal to the bit-width of e, the resulting value will be zero for unsigned types and the sign bit for signed types.
n must be non-negative.
```

I think this should a bug on Chisel side.

closed 2024-01-20 by mergify[bot]

SpriteOvO, 2024-02-17:
UPDATE: #3752 has been reverted, fixed more properly in #3824.

PR #3752 (2024-01-20): Shift right produce at least 1 width result
Fixes #3735.