chipsalliance/chisel #3746: Chisel v6.0.0 compile works, test fails

gonsolo, 2024-01-19:
Testing Chisel v6.0.0 with my small project resulted in the following error messages. Any ideas?
Version 5.1.0 works fine.

```
[info] compiling 3 Scala sources to /home/gonsolo/work/pepe/target/scala-2.13/test-classes ...
[info] DiffuseSpec:
[info] Diffuse
[info] bsdf.DiffuseSpec *** ABORTED ***
[info]   java.lang.NoClassDefFoundError: firrtl/ir/PartialConnect
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:2140)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:385)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$1(ChiselBridge.scala:155)
[info]   at scala.collection.immutable.List.map(List.scala:250)
[info] MultiplySpectrumSpec:
[info]   at scala.collection.immutable.List.map(List.scala:79)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:155)
[info]   ...
[info]   Cause: java.lang.ClassNotFoundException: firrtl.ir.PartialConnect
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:2140)
[info] bsdf.MultiplySpectrumSpec *** ABORTED ***
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:385)
[info]   java.lang.NoClassDefFoundError: firrtl/ir/PartialConnect
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$1(ChiselBridge.scala:155)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:2140)
[info]   at scala.collection.immutable.List.map(List.scala:250)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:385)
[info]   at scala.collection.immutable.List.map(List.scala:79)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:155)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   ...
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$1(ChiselBridge.scala:155)
[info]   at scala.collection.immutable.List.map(List.scala:250)
[info]   at scala.collection.immutable.List.map(List.scala:79)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:155)
[info]   ...
[info]   Cause: java.lang.ClassNotFoundException: firrtl.ir.PartialConnect
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:2140)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:385)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$1(ChiselBridge.scala:155)
[info]   at scala.collection.immutable.List.map(List.scala:250)
[info]   at scala.collection.immutable.List.map(List.scala:79)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:155)
[info]   ...
[info] MultiplySpec:
[info] bsdf.MultiplySpec *** ABORTED ***
[info]   java.lang.NoClassDefFoundError: firrtl/ir/PartialConnect
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:232)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$8(ChiselBridge.scala:227)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:2140)
[info]   at scala.collection.immutable.Vector1.map(Vector.scala:385)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:227)
[info]   at chiseltest.simulator.ChiselBridge$.convert(ChiselBridge.scala:167)
[info]   at chiseltest.simulator.ChiselBridge$.$anonfun$convert$1(ChiselBridge.scala:155)
[info]   at scala.collection.immutable.List.map(List.scala:250)
[info]   at scala.collection.immutable.List.map(List.scala:79)
[... 62 lines cut ...]
[error] The test class loader accesses the project dependency classes using the cached project dependency classloader.
[error] With this approach, class loading may fail under the following conditions:
[error] 
[error]  * Dependencies use reflection to access classes in your project's classpath.
[error]    Java serialization/deserialization may cause this.
[error]  * An open package is accessed across layers. If the project's classes access or extend
[error]    jvm package private classes defined in a project dependency, it may cause an IllegalAccessError
[error]    because the jvm enforces package private at the classloader level.
[error] 
[error] These issues, along with others that were not enumerated above, may be resolved by changing the class loader layering strategy.
[error] The Flat and ScalaLibrary strategies bundle the full project classpath in the same class loader.
[error] To use one of these strategies, set the  ClassLoaderLayeringStrategy key
[error] in your configuration, for example:
[error] 
[error] set root / Test / classLoaderLayeringStrategy := ClassLoaderLayeringStrategy.ScalaLibrary
[error] set root / Test / classLoaderLayeringStrategy := ClassLoaderLayeringStrategy.Flat
[error] 
[error] See ClassLoaderLayeringStrategy.scala for the full list of options.
[info] Run completed in 899 milliseconds.
[info] Total number of tests run: 0
[info] Suites: completed 0, aborted 3
[info] Tests: succeeded 0, failed 0, canceled 0, ignored 0, pending 0
[info] *** 3 SUITES ABORTED ***

```

jackkoenig, 2024-01-19:
This is a binary compatibility issue, please see Chisel's versioning policy: https://www.chisel-lang.org/docs/appendix/versioning

Basically, binary compatibility breaks between major versions, so ChiselTest compiled for Chisel 5.x will not work with Chisel 6.x. Kevin has indicated intent to release ChiselTest for Chisel 6: https://github.com/ucb-bar/chiseltest/issues/699.

closed 2024-01-19 by jackkoenig

gonsolo, 2024-01-20:
chiseltest sbt publishLocal needs the following patch:
```
diff --git a/src/main/scala/treadle2/stage/phases/GetFirrtlAst.scala b/src/main/scala/treadle2/stage/phases/GetFirrtlAst.scala
index 388846a..ed856f9 100644
--- a/src/main/scala/treadle2/stage/phases/GetFirrtlAst.scala
+++ b/src/main/scala/treadle2/stage/phases/GetFirrtlAst.scala
@@ -38,7 +38,7 @@ object GetFirrtlAst extends Phase {
     def handleFirrtlFile(): Option[AnnotationSeq] = {
       annotationSeq.collectFirst { case FirrtlFileAnnotation(fileName) => fileName } match {
         case Some(fileName) =>
-          val file = io.Source.fromFile(fileName)
+          val file = scala.io.Source.fromFile(fileName)
           val text = file.mkString
           file.close()
```

firrtl2 needs the following patch:
```
diff --git a/build.sbt b/build.sbt
index 895191b0..ffe0d80c 100644
--- a/build.sbt
+++ b/build.sbt
@@ -4,7 +4,7 @@ enablePlugins(SiteScaladocPlugin)
 
 lazy val commonSettings = Seq(
   organization := "edu.berkeley.cs",
-  scalaVersion := "3.2.2",
+  scalaVersion := "2.13.12",
   crossScalaVersions := Seq("2.13.10", "3.2.2")
 )
 
@@ -100,8 +100,7 @@ lazy val docSettings = Seq(
   Compile / doc / scalacOptions ++= Seq(
     // ANTLR-generated classes aren't really part of public API and cause
     // errors in ScalaDoc generation
-    "-skip-packages",
-    "firrtl2.antlr",
+    "-skip-packages:firrtl2.antlr",
     "-Xfatal-warnings",
     "-feature",
     "-diagrams",
```

After that my project fails with:

> [info]   java.lang.NoClassDefFoundError: chisel3/internal/firrtl/Width

From my point of view this issue is far from completed.
My wishlist:
1. Before releasing a new Chisel version, keep at least chisel-template and chiseltest working. (Alternatively: How are you testing at SiFive and Berkeley?)
2. At least keep chiseltest *compiling*.
3. Similarly keep firrtl2 compiling.
4. Don't fork an old compiler just.
4a. Using Scala 3 makes the forked compiler incompatible to the rest of Chisel/chiseltest/...

As for now I'm staying at 5.1.0, and I hope you consider not breaking external projects in the future.

kiniry, 2024-02-20:
Love your comment's wishlist number 1 in particular, @gonsolo....

Releasing Chisel 6 without having any structured means by which to perform assurance is gonzo.  CC @schoeberl (Hi Martin!)

schoeberl, 2024-02-22:
Hi Joseph (@kiniry), what is your interest in Chisel?

seldridge, 2024-02-27:
Hi all. Sorry about the breakages with ChiselTest. ChiselTest, from the perspective of the Chisel developers, was not intended to be used beyond Chisel 3.6. ChiselSim was built as a lightweight, maintained replacement in Chisel 5. _Please migrate to ChiselSim_. More information about the background of this issue and a migration guide is now provided here: http://www.chisel-lang.org/docs/appendix/migrating-from-chiseltest.

If there are features you want/need with ChiselSim (e.g., `enqueueSeq`/`dequeueSeq` or `fork`/`join`), please open issues or consider contributing these features to the project.

The [`chipsalliance/chisel-template`](https://github.com/chipsalliance/chisel-template) has been migrated to use ChiselSim.

kiniry, 2024-02-27:
Hey @schoeberl!  Remember me from DTU?  We hardly got to talk while I was working there for a year before joining [Galois](https://galois.com/) ten years (!) ago.

We have used Chisel at Galois on several Government-funded R&D projects I have run for the past eight years or so, including taping out several interesting ASICs.  E.g., we were responsible for the six SoCs (plus OSs, development toolchains, improvements to clang to support RISC-V, and much more) provided by the Government to [DARPA SSITH](https://www.darpa.mil/program/ssith) performers, and we simulated those SoCs in AWS for [DARPA FETT](https://fett.darpa.mil/), for which we built the entire infrastructure.

We are currently using Chisel on the [DARPA AISS[(https://www.darpa.mil/program/automatic-implementation-of-secure-silicon) program, as we are developing a product line of hardware security engines.  Thus, I need to be able to automatically generate model-based test benches and verification benches from our [RDE](https://galois.com/services/rigorous-digital-engineering/) (formally-grounded) designs/models.  

While we can go from RDE models to SVA-based benches (or, yuck, UVM), I'd rather push toward stronger integration of Chisel-centric assurance in this project, thus my nudge on this issue.