chipsalliance/chisel #3767: [Panama] Request to bind InstanceGraph API from CIRCT.

sequencer, 2024-01-25:
**Type of issue**: Feature Request

After reviewing the current OM C-API. I suggest introducing the `InstanceGraph` Binding from CIRCT. It should expose the `InstanceGraph` result to Scala.

Basically the current OM C-API is designed for Diplomatic-based design flow, which only need to query OM at top module, but waive all OM at inner layer. This does work under the magic of diplomacy AutoBundle and instantiate API, but... for most of chisel end users, it's less friendly...
 
IMO, for most of use cases, users just wanna add some annotations on the IR(e.g. retiming, clock domain, power domain). We can always use something like boring utils to bore them to Top for working around, but we can make it simpler...

An example Scala API can be represent below, for more efficiency operations, we can expose a bottom layer API in the future.
```
package panamalib {
  package instancegraph {
    case class Module(name: String, children: Seq[InstanceNode], instances: Seq[Instance])
    case class Instance(module: Module, father: Option[Module])
  }
}
```
After exposing this API, we can traverse the Modules in any MLIR circuit, and query OM inside any modules.

SpriteOvO, 2024-02-03:
Implemented in #3798.

sequencer, 2024-02-03:
Reopen it since #3798 only provide a Binding w/o user-level API.