chipsalliance/chisel #3884: The default C++ Standard flags in svsim is outdated for verilator

Emin017, 2024-02-28:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
Hi All,  I'm using chisel's svsim and ChiselSim to test my modules, but when I updated to verilator v5.022, the svsim tests would fail. Even using the examples provided in the [documentation](https://www.chisel-lang.org/docs/appendix/migrating-from-chiseltest) can lead to such failures. I just copied the example code and started running the ChiselSim test, and I get this output :
<details>

<summary>ChiselSim Output</summary>

```
make: Entering directory '/tmp/chisel3.simulator.EphemeralSimulator_275139@archlinux_9734783782933421822/workdir-default'
ls . | grep -v Makefile | grep -v execution-script.txt | xargs rm -rf
\
/usr/local/bin/verilator \
	'--cc' \
	'--exe' \
	'--build' \
	'-o' \
	'../simulation' \
	'--top-module' \
	'svsimTestbench' \
	'--Mdir' \
	'verilated-sources' \
	'-CFLAGS' \
	'-std=c++11 -I/tmp/chisel3.simulator.EphemeralSimulator_275139@archlinux_9734783782933421822/workdir-default -DSVSIM_ENABLE_VERILATOR_SUPPORT' \
	'../primary-sources/MyModule.sv' 'testbench.sv' '../generated-sources/c-dpi-bridge.cpp' '../generated-sources/simulation-driver.cpp'
make[1]: Entering directory '/tmp/chisel3.simulator.EphemeralSimulator_275139@archlinux_9734783782933421822/workdir-default/verilated-sources'
ccache clang++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++11 -I/tmp/chisel3.simulator.EphemeralSimulator_275139@archlinux_9734783782933421822/workdir-default -DSVSIM_ENABLE_VERILATOR_SUPPORT   -Os -c -o c-dpi-bridge.o c-dpi-bridge.cpp
ccache clang++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++11 -I/tmp/chisel3.simulator.EphemeralSimulator_275139@archlinux_9734783782933421822/workdir-default -DSVSIM_ENABLE_VERILATOR_SUPPORT   -Os -c -o simulation-driver.o simulation-driver.cpp
In file included from simulation-driver.cpp:883:
In file included from ./VsvsimTestbench.h:11:
In file included from /usr/local/share/verilator/include/verilated.h:42:
/usr/local/share/verilator/include/verilatedos.h:265:3: error: "Verilator requires a C++14 or newer compiler"
# error "Verilator requires a C++14 or newer compiler"
  ^
In file included from simulation-driver.cpp:883:
In file included from ./VsvsimTestbench.h:11:
/usr/local/share/verilator/include/verilated.h:76:22: error: expected namespace name
using namespace std::literals;  // "<std::string literal>"s; see SF.7 core guideline
                ~~~~~^
In file included from simulation-driver.cpp:883:
In file included from ./VsvsimTestbench.h:11:
In file included from /usr/local/share/verilator/include/verilated.h:950:
/usr/local/share/verilator/include/verilated_types.h:266:24: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wreserved-user-defined-literal]
    return "triggered="s + (e.isTriggered() ? "true" : "false");
                       ^
                        
/usr/local/share/verilator/include/verilated_types.h:266:24: error: expected ';' after return statement
    return "triggered="s + (e.isTriggered() ? "true" : "false");
                       ^
                       ;
/usr/local/share/verilator/include/verilated_types.h:277:24: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wreserved-user-defined-literal]
    return "triggered="s + (e.isTriggered() ? "true" : "false");
                       ^
                        
/usr/local/share/verilator/include/verilated_types.h:277:24: error: expected ';' after return statement
    return "triggered="s + (e.isTriggered() ? "true" : "false");
                       ^
                       ;
/usr/local/share/verilator/include/verilated_types.h:1651:18: error: no member named 'exchange' in namespace 'std'; did you mean 'vlstd::exchange'?
        : m_objp{std::exchange(moved.m_objp, nullptr)} {}
                 ^~~~~~~~~~~~~
                 vlstd::exchange
/usr/local/share/verilator/include/verilatedos.h:642:3: note: 'vlstd::exchange' declared here
T exchange(T& obj, U&& new_value) {
  ^
In file included from simulation-driver.cpp:883:
In file included from ./VsvsimTestbench.h:11:
In file included from /usr/local/share/verilator/include/verilated.h:950:
/usr/local/share/verilator/include/verilated_types.h:1661:18: error: no member named 'exchange' in namespace 'std'; did you mean 'vlstd::exchange'?
        : m_objp{std::exchange(moved.m_objp, nullptr)} {}
[... 186 lines cut ...]
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2or3(ScalaTestRunner.java:43)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:26)

``` 
</details>

Someone else has already raised the same issue: https://github.com/chipsalliance/chisel/issues/3836. And I found out that this is because verilator added a [check](https://github.com/verilator/verilator/commit/e3e5f30ba1b02a202aff32e10c95e824e855a6df)  for the c++14 standard.

**What is the current behavior?**
ChiselSim will use [`-std=c++11`](https://github.com/chipsalliance/chisel/blob/c161fccdc2341201f4c240cb38654d75aaaceadd/svsim/src/main/scala/verilator/Backend.scala#L123C1-L123C33) by default, which will cause the ChiselSim to fail the verilator check.
**What is the expected behavior?**
The minimum c++ standard expected by verilator is c++14. So ChiselSim should use `-std=c++14`.

**Please tell us about your environment:**
- version: `6.0.0`
- verilator version: `Verilator 5.022 2024-02-24 rev v5.020-157-g2b4852048`
- OS: `Linux archlinux 6.7.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 23 Feb 2024 16:31:48 +0000 x86_64 GNU/Linux`
- g++ version: `g++ (GCC) 13.2.1 20230801`
- clang++ version: `clang version 16.0.6`

**Other Information**

In order to pass the check of the new version of verilator, I changed the default c++ standard to c++14. The changed code seems to work fine on my machine. And I raised this [PR#3876](https://github.com/chipsalliance/chisel/pull/3876) to fix the problem, but it doesn't seem to be reviewed (Please bear with me if I'm wrong). At the same time, I'm not sure it's a good solution to update the c++-standard flag in code. I haven't found any other solution yet.
Hopefully this bug will be fixed so I can continue to use svsim or ChiselSim for module testing :)

seldridge, 2024-02-28:
The change seems reasonable. I wasn't able to reproduce the error with Verilator 5.022 on macOS and clang 15. However, that's not indicative that this isn't a problem on other systems.

jackkoenig, 2024-02-28:
I can't figure out how to link to the specific version notes, but Verilator 5.020 (released 2024/01/01) says it requires C++14: https://verilator.org/guide/latest/changes.html.

seldridge, 2024-02-28:
Thanks for the patch. Calling this fixed now that #3876 is merged.

closed 2024-02-28 by seldridge