chipsalliance/chisel #3911: How to generate unoptimzed  SystemVerilog file when useing Chisel 6

qiaomufenlin, 2024-03-07:
When I use chisel 3.6，I can generate the no optimized verilog file (infact, i want the tool dont touch my defined signals ) with the ChiselStage option "-X" "mverilog"。I dont know how to do that  with Chisel 6.0。

seldridge, 2024-03-07:
You can run with `ChiselStage.emitSystemVerilog(new Foo, firtoolOpts = Array("-disable-opt"))` to turn off _most_ optimizations.

If your motivation for disabling optimizations is because you want to interact with the output Verilog in a predictable way, there are several approaches that will work better:

1. You can use `dontTouch` to block optimizations on a signal. _This is not advised for use to preserve the name and existence of a signal and one of the other two approaches is better._
2. You can mix-in the trait [`Public`](https://javadoc.io/doc/org.chipsalliance/chisel_2.13/latest/chisel3/Public.html) to modules to mark them as public which will cause their ports to be preserved. This is a Chisel 7 feature.
3. You can add ports of `Probe` type to cause public modules to produce Verilog macros that you can use to introspect the design. This is a Chisel 6 feature.

Both (2) and (3) are documented in the [FIRRTL ABI](https://github.com/chipsalliance/firrtl-spec/blob/main/abi.md).

An example showing (2) and (3) is below:

``` scala
//> using scala "2.13.12"
//> using repository sonatype-s01:snapshots
//> using lib "org.chipsalliance::chisel::7.0.0-M1+28-3feaf838-SNAPSHOT"
//> using plugin "org.chipsalliance:::chisel-plugin::7.0.0-M1+28-3feaf838-SNAPSHOT"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import chisel3.probe.{define, Probe, ProbeValue}
import circt.stage.ChiselStage

// Bar is marked "Public" to indicate that it cannot be optimized using
// information about its instantiating context.
class Bar extends RawModule with Public {
  val a = IO(Flipped(Bool()))
  val b = IO(Bool())
  // "c" is a read probe that exposes information about the internals of module
  // "Bar".
  val c = IO(Probe(Bool()))

  val _notA = !a
  b :<= !(_notA)

  define(c, ProbeValue(_notA))
}

class Foo extends RawModule with Public {
  val bar = Module(new Bar)
  bar.a :<= false.B
}

object Main extends App {
  println(
    ChiselStage.emitSystemVerilog(new Foo, firtoolOpts = Array("-strip-debug-info"))
  )
}
```

This produces the following Verilog:

``` verilog
// Generated by CIRCT firtool-1.67.0
module Bar(
  input  a,
  output b
);

  wire _notA_probe = ~a;
  assign b = a;
endmodule

module Foo();
  Bar bar (
    .a (1'h0),
    .b (/* unused */)
  );
endmodule

// ----- 8< ----- FILE "ref_Bar.sv" ----- 8< -----

// Generated by CIRCT firtool-1.67.0
`define ref_Bar_c _notA_probe
```

Both `Bar`'s exact port-level interface is preserved. Additionally, it's probe ports result in the creation of the file `ref_Bar.sv` which defines a macro `ref_Bar_c` which can be used to read the result of an internal signal (which would be optimized away without it).

If you need the dead code used to compute `ref_Bar_c` removed from the design, you can make this probe layer colored (a Chisel 7 feature) and hide the dead code in a SystemVerilog bind.

Using that approach:

``` scala
import chisel3._
import chisel3.probe.{define, Probe, ProbeValue}
import chisel3.layer.{block, Convention, Layer}
import circt.stage.ChiselStage

object Baz extends Layer(Convention.Bind)

// Bar is marked "Public" to indicate that it cannot be optimized using
// information about its instantiating context.
class Bar extends RawModule with Public {
  val a = IO(Flipped(Bool()))
  val b = IO(Bool())
  // "c" is a read probe that exposes information about the internals of module
  // "Bar".
  val c = IO(Probe(Bool(), Baz))

  val _notA = !a
  b :<= !(_notA)

  block(Baz) {
    define(c, ProbeValue(_notA))
  }
}

class Foo extends RawModule with Public {
  val bar = Module(new Bar)
  bar.a :<= false.B
}
```

You can then produce the following Verilog:

``` verilog
// Generated by CIRCT firtool-1.67.0
module Bar_Baz(
  input _notA
);

  wire _notA_probe = _notA;
endmodule

module Bar(
  input  a,
  output b
);

  assign b = a;
endmodule

module Foo();
  Bar bar (
    .a (1'h0),
    .b (/* unused */)
  );
endmodule

// ----- 8< ----- FILE "layers_Foo_Baz.sv" ----- 8< -----

// Generated by CIRCT firtool-1.67.0
`ifndef layers_Foo_Baz
`define layers_Foo_Baz
bind Bar Bar_Baz baz (
  ._notA (~a)
);
`endif // layers_Foo_Baz

// ----- 8< ----- FILE "ref_Bar.sv" ----- 8< -----

// Generated by CIRCT firtool-1.67.0
`define ref_Bar_c baz._notA_probe
```

Documentation on much of this (other than basic usage of layers) is currently missing. 🥲

qiaomufenlin, 2024-03-07:
thanks for your advice.   
But I want the tool to preserve the signal names defined in chisel. That will be very helpfull when debug with the waveforms。(When I used the chisel3，I writed a python script to process the verilog files generated by FIRRTL. The script remove the single names with "_" prehead. Than, I get the file that I want. haha..)

seldridge, 2024-03-07:
Ah. Try using `-preserve-values=all` in that case.

qiaomufenlin, 2024-03-08:
Yeah, that's very useful. Thank you very much!

closed 2024-03-08 by qiaomufenlin