chipsalliance/chisel #3933: <file>.v name appended to generated SV file when a blackbox is used

wkkuna, 2024-03-15:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

I came across this issue when bumping a project from Chisel 5.0 -> 6.2, I've tried to narrow it down as much as possible:

I use a Blackbox feature to import a design in Verilog to Chisel via `Blackbox with HasBlackBoxResource`, I then interface it in another module and `emitSystemVerilogFile`.

The implemented classes:

```
package AIG

import chisel3._
import chisel3.util.HasBlackBoxResource
import circt.stage.ChiselStage

class AcceleratorBlackbox extends BlackBox with HasBlackBoxResource {
  override val desiredName = "fpga_isp"
  val io = IO(new Bundle {
    val sys_clk = Input(Clock())
    val sys_rst = Input(Reset())

    val raw_payload_data = Input(UInt(32.W))
    val raw_valid = Input(Bool())
    val raw_ready = Output(Bool())
    val raw_first = Input(Bool())
    val raw_last = Input(Bool())

    val rgb_payload_data = Output(UInt(32.W))
    val rgb_valid = Output(Bool())
    val rgb_ready = Input(Bool())
    val rgb_first = Output(Bool())
    val rgb_last = Output(Bool())

    val bgr = Input(Bool())
    val pattern = Input(UInt(2.W))
    val busy = Output(Bool())
    val algorithm = Input(UInt(3.W))

    val rows = Input(UInt(13.W))
    val cols = Input(UInt(13.W))
  })
  addResource("fpga_isp.sv")
}

class AcceleratorModule extends Module {  
  val accelerator = Module(new AcceleratorBlackbox)
  val io = IO(new Bundle {

    val raw_payload_data = Input(UInt(32.W))
    val raw_valid = Input(Bool())
    val raw_first = Input(Bool())
    val raw_last = Input(Bool())

    val rgb_ready = Input(Bool())

    val bgr = Input(Bool())
    val pattern = Input(UInt(2.W))
    val algorithm = Input(UInt(3.W))

    val rows = Input(UInt(13.W))
    val cols = Input(UInt(13.W))
  })

   accelerator.io.sys_clk := clock
   accelerator.io.sys_rst := reset

  accelerator.io.raw_payload_data := io.raw_payload_data
  accelerator.io.raw_valid := io.raw_valid
  accelerator.io.raw_first := io.raw_first
  accelerator.io.raw_last := io.raw_last

  accelerator.io.rgb_ready := io.rgb_ready

  accelerator.io.bgr := io.bgr
  accelerator.io.pattern := io.pattern
  accelerator.io.algorithm := io.algorithm
  
  accelerator.io.rows := io.rows
  accelerator.io.cols := io.cols
}

object Driver extends App {
  ChiselStage.emitSystemVerilogFile(new AcceleratorModule)
}
```

I use the `sbt` and run with `sbt "runMain AIG.Driver"`

Here's a resource file I'm using:
[fpga_ispSV.log](https://github.com/chipsalliance/chisel/files/14616026/fpga_ispSV.log)

And here's the output SystemVerilog file:
[AcceleratorModuleSV.log](https://github.com/chipsalliance/chisel/files/14615962/AcceleratorModuleSV.log)

**What is the current behavior?**

The `./fpga_isp.v` line is appended at the end of `AcceleratorModule.sv` file.

**What is the expected behavior?**

The name of the included resource to not be appended to the end of the file as it causes an error with the later used SystemVerilog file.
I've searched the specification but I didn't find it to be a legal expression.

**Please tell us about your environment:**
    - chisel version: `6.2.0`
    - scala version: `2.13.12`
    - firtool version: `1.68.0`
    - sbt version: `1.6.2`
    - `java-11-openjdk`
    - OS: `Linux Terassen 6.4.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 27 Jul 2023 22:02:18 +0000 x86_64 GNU/Linux`

seldridge, 2024-03-15:
The single-file output is not super well supported and is more intended for testing. What's happening is that this is outputting the contents of all files into one file. Can you try with:

```
ChiselStage.emitSystemVerilogFile(new AcceleratorModule, Array("--split-verilog"))
```

wkkuna, 2024-03-15:
Thanks for the explanation.
The issue doesn't persist with `"--split-verilog"` option and the files are synthesisable.

CSharperMantle, 2024-05-26:
I am having this issue as well when defining an inline BlackBox with Chisel 7.0.0-M1. My BlackBox is named `MemuBlackBox`, and the final single-file output as these lines appended:
```verilog
// ----- 8< ----- FILE "./MemuBlackBox.sv" ----- 8< -----

// Generated by CIRCT firtool-1.66.0

module MemuBlackBox(
  ...
);
  ...
endmodule
  

// ----- 8< ----- FILE "firrtl_black_box_resource_files.f" ----- 8< -----

MemuBlackBox.sv
```
which is consistent with @wkkuna's observation.

However, after implementing @seldridge's solution (appending `--split-verilog`), Chisel no longer produces my top-level module. Suppose my top module is named `Top`, then the build artifact will contain an *empty* file named `Top.sv`.

Is there a way to fix this?

IveanEx, 2024-08-01:
I encountered the same problem. Really think it is a bug and should be fixed.

IveanEx, 2024-08-01:
> The single-file output is not super well supported and is more intended for testing. What's happening is that this is outputting the contents of all files into one file. Can you try with:
> 
> ```
> ChiselStage.emitSystemVerilogFile(new AcceleratorModule, Array("--split-verilog"))
> ```

I encountered the same problem... Especially when using HasBlackBoxInline and setInline, the "--split-verilog" seems not effectively split the blackbox to a seperated file, so I can only a script to delete it

pmjhhhh, 2024-12-18:
> I am having this issue as well when defining an inline BlackBox with Chisel 7.0.0-M1. My BlackBox is named `MemuBlackBox`, and the final single-file output as these lines appended:我在使用 Chisel 7.0.0-M1 定义内联 BlackBox 时也遇到了这个问题。我的 BlackBox 名为 `MemuBlackBox` ，最终的单文件输出附加以下几行：
> 
> ```verilog
> // ----- 8< ----- FILE "./MemuBlackBox.sv" ----- 8< -----
> 
> // Generated by CIRCT firtool-1.66.0
> 
> module MemuBlackBox(
>   ...
> );
>   ...
> endmodule
>   
> 
> // ----- 8< ----- FILE "firrtl_black_box_resource_files.f" ----- 8< -----
> 
> MemuBlackBox.sv
> ```
> 
> which is consistent with @wkkuna's observation.这与 的观察是一致的。
> 
> However, after implementing @seldridge's solution (appending `--split-verilog`), Chisel no longer produces my top-level module. Suppose my top module is named `Top`, then the build artifact will contain an _empty_ file named `Top.sv`.但是，在实施 的解决方案（附加 `--split-verilog` ）之后，Chisel 不再生成我的顶级模块。假设我的顶级模块名为 `Top` ，那么构建工件将包含一个名为 `Top.sv` 的空文件。
> 
> Is there a way to fix this?有办法解决这个问题吗？
> 
> Since I have not come to a solution of this issue, I've written a script for manually stripping all .f files from generated output. The following script overwrites each files specified in cmdline with stripped content.由于我还没有找到此问题的解决方案，因此我编写了一个脚本来手动从生成的输出中删除所有 .f 文件。以下脚本使用剥离的内容覆盖 cmdline 中指定的每个文件。
> 
> ```python
> #!/usr/bin/env python3
> 
> import re
> from argparse import ArgumentParser
> 
> RE_BANNER = re.compile(r"^// ----- 8< ----- FILE \"((?:.+)\.(?:.+))\" ----- 8< -----$")
> RE_FILE_F = re.compile(r"^.+\.f$")
> 
> 
> def match_banner(line: str) -> str | None:
>     match RE_BANNER.match(line):
>         case None:
>             return None
>         case m:
>             return m.group(1)
> 
> 
> parser = ArgumentParser(
>     description="Remove .f content from SystemVerilog files generated by Chisel"
> )
> parser.add_argument(
>     "filenames",
>     metavar="FILE",
>     type=str,
>     nargs="+",
>     help="SystemVerilog files to process",
> )
> args = parser.parse_args()
> 
> for filename in args.filenames:
>     with open(filename, "rt") as f:
>         lines = f.readlines()
> 
>     with open(filename, "wt") as f:
>         in_f_file = False
>         for banners in enumerate(map(match_banner, lines)):
>             if banners[1] is None:
>                 pass
>             elif RE_FILE_F.match(banners[1]) is None:
>                 in_f_file = False
>             else:
>                 in_f_file = True
>             if not in_f_file:
>                 print(lines[banners[0]], end="", file=f)
> ```
I made some unexpected discoveries. To solve issues like "The ./fpga_isp.v line is appended at the end of the AcceleratorModule.sv file," I added `--split-verilog` and `-o=build` to the firtoolOptions. As a result, the top module generated was empty, just like in your example, "then the build artifact will contain an empty file named Top.sv."

Then, I found that the reason was that my Makefile already specified the output folder as build:
`mill -i $(PRJ).runMain Elaborate --target-dir $(BUILD_DIR)`
After I removed `--target-dir $(BUILD_DIR)`, the top module was generated correctly. So I suspect that this was causing the empty top module to overwrite the correct top module.

jnbrq, 2025-04-07:
Is there a workaround for this issue?

closed 2025-04-07 by seldridge

Emin017, 2025-04-08:
> I made some unexpected discoveries. To solve issues like "The ./fpga_isp.v line is appended at the end of the AcceleratorModule.sv file," I added `--split-verilog` and `-o=build` to the firtoolOptions. As a result, the top module generated was empty, just like in your example, "then the build artifact will contain an empty file named Top.sv."
> 
> Then, I found that the reason was that my Makefile already specified the output folder as build:
> 
> ```
> BUILD_DIR = ./build
> verilog:
> 	mkdir -p $(BUILD_DIR)
> 	mill -i $(PRJ).runMain Elaborate --target-dir $(BUILD_DIR)
> ```
> 
> After I removed `--target-dir $(BUILD_DIR)`, the top module was generated correctly, and the empty top module was created in my current working directory. So, I suspect that the empty top module generated later overwrote the correct top module, causing the issue. Changing the output directory resolves the problem.

@Dev04LYY Are you using `emitSystemVerilog` instead of `emitSystemVerilogFile`? I think this might related: https://github.com/chipsalliance/chisel/issues/3973

commit 3b179e36 (2025-04-07): Add --split-verilog to emitSystemVerilogFile (#4867)

Change the behavior of `ChiselStage.emitSystemVerilogFile` to add
`--split-verilog`.  Without this, a user can run into a situation
where (rightly), they get non-Verilog output in the single file output.
Single file output from CIRCT is only intended to be used for testing.

Instead of telling users that they have to add "--split-verilog", just add
it by default with no way to turn it off.  I don't see a reason why a user
_should_ be able to turn this off as the "off" mode is only for testing
and this testing can be just as easily done with [... cut]