chipsalliance/chisel #3939: [Feature Request] Raise an error if a property path is not hierarchically underneath the same owning module

SpriteOvO, 2024-03-15:
Minimum reproducible example

```scala
import chisel3._
import chisel3.properties._
import chisel3.experimental.hierarchy.{instantiable, public, Definition, Instance}

@instantiable
class OM extends Class {
  val path = IO(Output(Property[Path]()))
  @public val pathIn = IO(Input(Property[Path]()))
  path := pathIn
}

class Sub extends Module {}

class Top extends Module {
  val omDef = Definition(new OM)
  val omIns = Instance(omDef)
  val sub = Module(new Sub)
  omIns.pathIn := Property(sub)
}

class TopWrapper extends Module {
  val top = Module(new Top)
}
```

CIRCT Firtool reports an error `unable to resolve path relative to owning module "Top"` for the outputs of `emitCHIRRTL(new TopWrapper)`, but works fine with the outputs of `emitCHIRRTL(new Top)`.

Thanks to @mikeurbach for explaining further information about that on Discord

> The rule is the paths in the OM dialect need to be hierarchically underneath the same "owning module" as the entities they target. https://github.com/llvm/circt/blob/6914bc6e0d11468c3fb11dfda74091f6e2e0306d/include/circt/Dialect/FIRRTL/OwningModuleCache.h#L18-L22.

However, the error will be reported by CIRCT Firtool, it appeared a bit late. It would be better if there is a possibility to check and report a more friendly error on Chisel.

CC @sequencer

sequencer, 2024-03-24:
This is tricky: seems Class/Object and Path API is happier with D/I API.

Think about for each OM Class we instantiate under a Module, we cannot use `Path` API to annotate `Path` beneath it, since it's not:
> hierarchically underneath the same "owning module" as the entities they target

I'm not sure if this is a bug? cc: @seldridge, @jackkoenig, @azidar, If it is, should it be fixed in Chisel or in CIRCT? And will this fixing introduce problems in dedup?

In our team, we want to have use the API from OM and D/I for better metadata and deduplication at chisel elaboration time. I thought they are decoupled, so we can only firstly introduce OM to us.

However the problem seems to be: OM is depending on D/I now. while it's tricky for our team using IDEA Intellij for Chisel development, and [InstantiableMacro](https://github.com/chipsalliance/chisel/blob/main/macros/src/main/scala/chisel3/internal/InstantiableMacro.scala) make IDEA extreme unhappy:( Thus we are blocked by the `chisel-idea-plugin` which may use [SyntheticMembersInjector](https://github.com/JetBrains/intellij-scala/blob/idea233.x/scala/scala-impl/src/org/jetbrains/plugins/scala/lang/psi/impl/toplevel/typedef/SyntheticMembersInjector.scala) to insert public `val` to `Instance` enclosure.

seldridge, 2024-03-25:
This seems to be the same conceptual problem as input probes where allowing input probes results in circuits which may have no legal lowering because a path is context sensitive (without resorting to duplication or simplifying probes to remove u-turn connections).

Input probes historically tried really hard to check if they were legal. However, we've since walked back that support in the FIRRTL spec. I expect that this is the most reasonable thing to do here, too.

Is the circuit above from a real design or is it handcrafted?

sequencer, 2024-03-25:
Not actually, the problem is:
> The rule is the paths in the OM dialect need to be hierarchically underneath the same "owning module" as the entities they target. https://github.com/llvm/circt/blob/6914bc6e0d11468c3fb11dfda74091f6e2e0306d/include/circt/Dialect/FIRRTL/OwningModuleCache.h#L18-L22.

Which means: `Path Property` can only be used at two situations:
- one OM at Top, use Path pointing to any hierarchy value in the Module.
- using D/I, OM at each Top of Definition, point to any hierarchy value underneath the Defination.

Our use case in T1 are actually the third version: we don't use D/I, but create OM at intermediate module. But it won't work, because:
> The rule is the paths in the OM dialect need to be hierarchically underneath the same "owning module" as the entities they target.

seldridge, 2024-03-25:
Ah, I see... @mikeurbach: do you have any thoughts on how this usage may be wrong?

mikeurbach, 2024-03-25:
> The rule is the paths in the OM dialect need to be hierarchically underneath the same "owning module" as the entities they target.

This is indeed the rule. For context, this is for multiple instantiation. If you have some hierarchy Foo with OM in it, and you instantiate it one or more times, CIRCT needs to be able to prepend the hierarchies where Foo was instantiated in front of any Paths within Foo. If the paths point to things that are hierarchically above Foo, this won't work. In other words, if you use Paths within some hierarchy, they need to be encapsulated by that hierarchy.

> Which means: Path Property can only be used at two situations:
> 
> * one OM at Top, use Path pointing to any hierarchy value in the Module.
> * using D/I, OM at each Top of Definition, point to any hierarchy value underneath the Defination.

I would untangle this a bit, it is not one or the other, and whether D/I is used or the unsafe APIs are used is orthogonal. As long as the rule above is respected, there is flexibility. So if you want to create OM at some intermediate module Foo in T1, the Paths must refer to things within that intermediate module Foo. If you want to create Paths that refer to things that are above Foo, you need to create those Paths in a common ancestor of the things they refer to.

One way to handle this is to have multiple Class/Object instances at different layers of the hierarchy. You can then make those Objects outputs that come out the intermediate hierarchies as output Property[ClassType] ports. Depending on how the generator is written you could using BoringUtils.bore to pull Object references from different levels of the hierarchy to be outputs at some top-level module.