chipsalliance/chisel #3942: `PanamaCIRCTConverter` emits incorrect firrtl for negative `SInt` literals.

OceanS2000, 2024-03-18:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
When developing integration tests for it (#3936), I noticed the following case will fail:

```scala
class WireAndReg extends Module {
  val r = IO(Input(Bool()))
  val o = IO(Output(UInt(2.W)))

  val o_next = RegInit(false.B)
  val flip = Reg(Bool())
  val magic = Wire(SInt(8.W))

  o := o_next && magic(7)
  o_next := flip
  flip := flip ^ r
  magic := -42.S
}

println(lit.utility.panamaconverter.firrtlString(new WireAndReg))
```

**What is the current behavior?**
For Wire `magic`, PanamaConverter will emit `connect magic, pad(SInt<7>(22), 8)`. For some reason, the sign bit is truncated, and in Verilog generation the whole circuit will be optimized out.

**What is the expected behavior?**
For reference, `circt.stage.ChiselStage.emitCHIRRTL` emits `connect magic, asSInt(UInt<7>(0h56))`.

**Please tell us about your environment:**

`nix develop` shell on HEAD as of 31b222e4445b1d6c58321e1d4d1752b345e02326.

**What is the use case for changing the behavior?**

Give correct negative `SInt`s.

OceanS2000, 2024-03-18:
@SpriteOvO could probably look into this?

SpriteOvO, 2024-03-18:
I'll look into it tonight, thanks for report.

PR #3943 (2024-03-23): Fix incorrect width for negative `SInt` literals
Fixes #3942.

The reason of this bug is that the result of Scala's `BigInt.bitLength` excludes the sign bit.

Reference from Chisel

https://github.com/chipsalliance/chisel/blob/7b9c07c4e5fe4a8174b4f788cf798b3870039e7e/core/src/main/scala/chisel3/internal/firrtl/Converter.scala#L88-L92

I haven't figured out why Chisel always generates `asSInt(UInt())` for `SLit`. However, since `pad` is sign-extends this behavior is documented. I think the fixed version `connect magic, pad(SInt<7>(-42), 8)` and `connect magic, asSInt(UInt<7>(0h56))` should be e [... cut]
closed 2024-03-23 by chiselbot