chipsalliance/chisel #3962: svsim: Verilator `initial` statements are not handled correctly

ubfx, 2024-04-03:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
Build a chisel module with a `Mem`, which have the option to be randomly initialized. Use svsim to simulate, and try using the random initialization by setting `+define+ENABLE_INITIAL_MEM_` and `+define+RANDOMIZE_MEM_INIT`.

**What is the current behavior?**
A testbench is generated containing something like this:
testbench.sv:
```verilog
// Simulation
  import "DPI-C" context task simulation_body();
  initial begin
    simulation_body();
  end
```

However, all the (randomized) memory initialization also happens in initial blocks, e.g.
mymem.sv:
```verilog
 `ifdef ENABLE_INITIAL_MEM_
    reg [31:0] _RANDOM_MEM;
    initial begin	
      ...
```

The initial blocks are scheduled such that the one in the testbench is executed first, at least in my case. Since `simulation_body()` executes the entire simulation, the entire simulation runs before all the other `initial` blocks, therefore skipping any initialization.

**What is the expected behavior?**
The `simulation_body()` should run after all other initial blocks.

A quick fix, although not pretty, would be to wrap the `simulation_body()` call in `final begin .. end` instead of `initial begin .. end`, where it would be guaranteed to run after the initial blocks.

**Please tell us about your environment:**

Chisel 6.2.0

**Other Information**

**What is the use case for changing the behavior?**
It's quite important to have the randomization feature working because verilator doesn't support X values. Running the entire simulation before the initial blocks (and therefore any randomization) will hide bugs.

kayvan61, 2024-11-21:
Why is it a bad idea to have the simulation body in a final block? I have an out of tree change that does this, but don't know enough about the intricacies of systemverilog to know why its a bad idea.

ubfx, 2024-11-26:
> about the intricacies of systemverilog to know why its a bad idea.

One thing that comes to mind is that IIRC the order of execution of multiple `final` blocks is not well-defined. So if you have processes which print statistics at the end of the run, this can go wrong. Best example would be increasing an error counter in the main program logic and printing it in the final block - this would lead to "0 errors" being printed before the main logic even ran.

seldridge, 2025-01-14:
Sorry this took so long to fix. I have a candidate fix out in https://github.com/chipsalliance/chisel/pull/4614. This opts for an explicit state machine inside the generated testbench that will do start simulation in an always block after an initial block runs. This should avoid ordering problems with other initial blocks and avoid problems of any ordering of `final` blocks.

PR #4614 (2025-01-14): [svsim] Fix initial ordering issue
Fix a bug in svsim where the entire simulation is run inside an `initial` block from a generated testbench.  This creates undefined simulation ordering with `initial` blocks in the simulated modules.  Fix this by switching to an explicit state machine inside the testbench.

Fixes #3962.

#### Release Notes

- Fix initial ordering bug in svsim.
commit ab7c417e (2025-01-14): [svsim] Fix initial ordering issue (#4614)

Fix a bug in svsim where the entire simulation is run inside an `initial`
block from a generated testbench.  This creates undefined simulation
ordering with `initial` blocks in the simulated modules.  Fix this by
switching to an explicit state machine inside the testbench.

Fixes #3962.
closed 2025-01-14 by seldridge