chipsalliance/chisel #3972: Workspace path parsing fails on Windows

zhc7, 2024-04-05:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

1. clone chisel template
2. run `sbt test`
3. it fails

**What is the current behavior?**

I traced it down to this:
`EphemeralSimulator.makeSimulator` creates a Simulator with workspace path that starts with "C:\", then when parsing this temp dir at https://github.com/chipsalliance/chisel/blob/ce05462c239deec8aee0704efbd3a71e8b64ec49/svsim/src/main/scala/Workspace.scala#L32, it is mistakenly considered as a relative path which leads to errors.

**What is the expected behavior?**

Maybe we can find a way that is more friendly to Windows? For example add `|| path(1) == ':'`.

**Please tell us about your environment:**

 - version: `6.2.0`
 - OS: Windows with msys2 environment

**Other Information**

**What is the use case for changing the behavior?**

To run it on Windows.

Acutally I'm not sure if chisel is intended not to support Windows. I didn't see any warnings about this. If so, please tell me.

jackkoenig, 2024-04-05:
Thanks for the report!

Chisel is supposed to support Windows, although the level of support is a bit up in the air at the moment. We definitely support emitting Verilog, but it gets trickier when it comes to running simulations. Prior to Chisel 5.0.0 when we were still using the [Scala FIRRTL Compiler](https://github.com/chipsalliance/firrtl), we had an interpreter written in Scala ([Treadle](https://github.com/chipsalliance/treadle)) which made it easy to support simulations on Windows. Now, we rely on [Verilator](https://www.veripool.org/verilator/) to simulate our emitted Verilog and I believe Verilator's support for Windows is pretty limited. That being said, the Scala code in Chisel should not be the limiting factor here so we should fix this issue you're seeing. I also suspect Verilator should work on MSYS2 so that excuse doesn't apply 😉.

There is also movement toward having Chisel link against [CIRCT](https://github.com/llvm/circt) directly rather than just running the firtool binary (the later is published for Windows). Unfortunately, LLVM's build flow doesn't support building Windows DLLs so how we make the direct linking work on Windows is unclear. It may be the case that we just need to push Chisel Windows users to use WSL2 which should allow us to just use the Linux shared libraries.

A bit rambly, but hopefully this provides a little bit more context for where things are at. We would love someone who uses Windows to help improve Windows support 🙂

zhc7, 2024-04-06:
Thanks for replying!

WSL2 is probably the best work around. After resolving this issue locally, I found that there's still some subtle traps, including but not limited to: verilator 5.x does not work; path separator `\` should be replaced by  `/` in makefile; g++ provided by msys2 can't resolve `getline` in cpp, etc. It's much easier to just embrace wsl. Though, I'm still looking forward to running chisel natively on Windows. I'm happy to help if needed!

sequencer, 2024-04-06:
For those issue you described, I think maybe it’s a right time for chisel to deprecate native Windows support ;p

seldridge, 2025-03-04:
Yeah, that specific piece of code is pretty bad.

I'm not against more first-class windows support and we could start putting this in CI, even if it was Windows WSL2 testing.

If you'd like to take a stab at, go for it. I do not have access to a Windows machine and can't do any testing other than painfully bouncing things off CI.