chipsalliance/chisel #3973: EmitVerilog generate empty output file with firtool option `split-verilog`

unlsycn, 2024-04-06:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
```
  chisel3.emitVerilog(
    new Top,
    ("-td=/path/to/output") +: firrtlOpts,
    Seq(
      FirtoolOption("--split-verilog"),
      FirtoolOption("-o=." + getArg("targetDir"))
    )
  )
```

**What is the current behavior?**
The firrtl still seems to take input from firtool but get nothing and outputs the empty file since firtool has been output to the file.

**What is the expected behavior?**
EmitVerilog no longer outputs to file if the output is empty.

**Please tell us about your environment:**
    - version: `7.0.0-M1`
    - firtool version: `1.66.0`
    - OS: `Gentoo Linux`

**Other Information**

**What is the use case for changing the behavior?**
When we prefer to output by `split-verilog` (this should be the majority of cases)

seldridge, 2024-04-08:
`chisel3.emitVerilog` won't do what you want. That's a thin wrapper around `ChiselStage.emitSystemVerilog` which is where the main problem is. These functions that return strings do not work correctly with some options manually specified to `firtool` because they rely on these options _not_ being specified to `firtool` so that they can grab the stdout outputs and pull it back into the string that is returned. `--split-verilog` and `-o` are incompatible with these.

Generally, you either want to use `ChiselStage.emitSystemVerilog` if you want a `String`. Use `ChiselStage.emitSystemVerilogFile` with the `--split-verilog` option _passed to Chisel not `firtool`_ if you want multi-file output. There is no way to get both a string and file output.

unlsycn, 2024-04-08:
Thx a lot!

closed 2024-04-08 by unlsycn